adafruit/circuitpython #5103: Receive command from host computer via SerialPort

TanguyBoulard, 2021-08-06:
_Hi all,_

I am using {Adafruit Metro M0 Express + Serial communication + Adafruit Shileld Motor Shield V2 + Stepper motor}, and I'm programming it with Circuit Python.
I want to receive command transmitted from the computer to which the Metro is connected via USB. Then I can execute the right command and activate motors.

**CircuitPython Code:**
```
import time
import board
from adafruit_motor import stepper
from adafruit_motorkit import MotorKit
import supervisor

kit = MotorKit(i2c=board.I2C())

def FORWARD():
    for i in range(100):
        kit.stepper2.onestep(direction=stepper.FORWARD, style=stepper.SINGLE)
        time.sleep(0.01)

def BACKWARD():
    for i in range(100):
        kit.stepper2.onestep(direction=stepper.BACKWARD, style=stepper.SINGLE)
        time.sleep(0.01)

def EXECUTE():
    if value == 'forward':
            FORWARD()

    elif value == 'backward':
            BACKWARD()

while True:
    if supervisor.runtime.serial_bytes_available:
        value = input().strip()
        command = value.decode('utf-8')
        EXECUTE(command)
```

**PC code:**
```
import serial

port = serial.Serial('COM16', 115200)
if not port.isOpen():
    port.open()
port.flush()

command = b'forward\n\r'
port.write(command)

port.close()
```

But nothing happen!

dastels, 2021-08-06:
Try reversing the \r and \n.   The input() function might be looking for the windows terminator convention "\r\n"

TanguyBoulard, 2021-08-06:
Nope @dastels  it does not solve the issue.
Might the issue is located here : `command = value.decode('utf-8')`
I do even know how to debug it.

dhalbert, 2021-08-06:
Try printing the data that you are receiving, and make sure that the COM port is correct.

We don't do this kind of support via GitHub, as it is not a bug or enhancement request, but a support issue. Too many people get email each time a comment is made here. We can move this to discord: https://adafru.it/discord or to the forums: https://forums.adafruit.com/viewforum.php?f=60. Let us know which you choose. I am closing this issue for now.

closed 2021-08-06 by dhalbert

dastels, 2021-08-06:
Yeah. I dug through the related code and it is looking for "\n" as the line terminator.

I've had success using sys.stdin.read(1) to get one character at a time from the serial/usb connection.

dhalbert, 2021-08-06:
I see you have already posted in the forums: https://forums.adafruit.com/viewtopic.php?f=60&t=182014. Let's continue the discussion there.