adafruit/circuitpython #5105: Include Board_ID in the boot_out.txt for Circup to use to link to CP.org

askpatrickw, 2021-08-06:
As discussed in the Circup repo https://github.com/adafruit/circup/issues/113

Having a way for circup to determine a link to the CP.org page would allow circup to direct users where exactly to go when there is a newer version of CP detected for their board.

As per Neradoc "the process would be to add the board ID somewhere as a define in the core, whether automatically generated or manually, then add it to the boot_out.txt which is as simple as adding the define name to this line:" 

https://github.com/adafruit/circuitpython/blob/be2342f32f80a0a2ba6839333847b1fdb495b439/py/makeversionhdr.py#L110

Scott had also mentioned he'd like to see this available in `board possibly as `board.ID`.

NathanY3G, 2021-08-08:
I would like to volunteer to take a look at adding this :-)

Neradoc, 2021-08-09:
There's `CIRCUITPY_BOARD_ID` already defined (only used on rp2040 for picotool information). `MICROPY_FULL_VERSION_INFO` is also the string printed to the REPL, and I want the resulting string to be easily parsable.

I did some tests on a couple of boards. Adding it at the end of `MICROPY_FULL_VERSION_INFO` like this `" (" CIRCUITPY_BOARD_ID ")"` takes 68 bytes on Arduino Nano 2040 and 64 bytes on a Gemma.

I also tried printing it to `boot_out.txt` from main.c without changing `MICROPY_FULL_VERSION_INFO` (and therefore the string printed to the REPL), on the second line like this:
`mp_hal_stdout_tx_str("\r\n" "Board ID:" CIRCUITPY_BOARD_ID "\r\n");`
And that seems to take 84 bytes on an Arduino Nano 2040, 40 bytes on a Gemma.

Adding an entry to `board` seems to require adding an entry in every `pins.c`, or is there an other way ?

dhalbert, 2021-08-12:
> @tannewt Is Neradoc correct that "Adding an entry to board seems to require adding an entry in every `pins.c`?"

It would, but how about using the existing `os.uname()`, such as `os.uname().machine`?

Currently:
```py
>>> os.uname()
(sysname='rp2040', nodename='rp2040', release='7.0.0', version='7.0.0-alpha.6-3-g5d6706ce8-dirty on 2021-08-12', machine='Adafruit QT Py RP2040 with rp2040')
```
We could add the board ID to `machine`, maybe putting it parentheses or something like that.

`os.uname()` exists in CPython on Linux and MacOS. Its `machine` is not that informative:
```py
>>> os.uname()
posix.uname_result(sysname='Linux', nodename='tuna', release='5.11.0-25-generic', version='#27~20.04.1-Ubuntu SMP Tue Jul 13 17:41:23 UTC 2021', machine='x86_64')
```

There is also `platform`, which is in CPython, but not in CircuitPython or MicroPython.

askpatrickw, 2021-08-13:
For Circup: As long as its nicely parsable in boot.txt it doesn't matter.
For CP: My opinion is that `board` makes more sense. It feels like the place to look for it to me.

If the reason for not doing doing it in board is because of needing to edit so many `pins.c` files.. I'm happy to help with that tedium in a follow-up PR if someone does the actual C work.

tannewt, 2021-08-24:
Anyone have a PR/branch for this? I feel like I've seen one.

What do we need to get it in?