adafruit/circuitpython #5111: board.LED not defined for Grand Central M4 Express

NathanY3G, 2021-08-08:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.5-140-g9a6b11806 on 2021-08-08; Adafruit Grand Central M4 Express with samd51p20
```

### Code/REPL

```python
import board
from digitalio import DigitalInOut, Direction
led = DigitalInOut(board.LED)
```

### Behavior

The following error is displayed:

```
AttributeError: 'module' object has no attribute 'LED'
```

anecdata, 2021-08-08:
The Learn Guide (https://learn.adafruit.com/adafruit-grand-central/pinouts) points out that the silk for the user LED for `D13` is `L`, but there is not a `board.L` either that I can see (https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/grandcentral_m4_express/pins.c).

NathanY3G, 2021-08-08:
Thanks for the information @anecdata and file reference. Since it looks like it should be a one-liner I will take the liberty of creating a MR myself.

FoamyGuy, 2021-08-08:
It looks to me like the Metro M0 and Metro M4 devices are in the same boat as this without `LED` or `L`

I'm going to take a shot at adding it to those devices.

NathanY3G, 2021-08-08:
I have just seen your reply @FoamyGuy! Thank you both for such quick replies :-)

I have already created a MR for the GCM4: https://github.com/adafruit/circuitpython/pull/5112 if anybody has time to take a look.

FoamyGuy, 2021-08-08:
actually, I'm not sure how I missed it in both, but the Metro M0 and M4 do already have them. 

Anyhow, thanks for adding that to the Grand Central build.

PR #5112 (2021-08-09): Add board.LED for Grand Central M4 Express
Fixes #5111
commit a15ac65f (2021-08-09): Add board.LED for Grand Central M4 Express
Fixes #5111
closed 2021-08-09 by microDev1