adafruit/circuitpython #5118: CPB CircuitPython doesn't behave as expected with potentiometer

gallaugher, 2021-08-09:
### CircuitPython version

```python
Adafruit CircuitPython 6.3.0 on 2021-06-01; Adafruit Circuit Playground Bluefruit with nRF52840
```

### Code/REPL

```python
# Note: same code as in this tutorial: https://learn.adafruit.com/make-it-change-potentiometers/circuitpython
import time
import board
from analogio import AnalogIn
potentiometer = AnalogIn(board.A1)  # potentiometer connected to A1, power & ground
while True:
    print((potentiometer.value,))      # Display value
    time.sleep(0.25)                   # Wait a bit before checking all again
```

### Behavior

When running this code on a CPB, the potentiometer maxes out as expected, but for the last about 1/10th, it retreats (note that image below). When running on other boards (have tried on a Feather RP2040 and a CPX), the behavior is as expected, maxes out, and no "dip". You can see the CPX, below, but same behavior is shown for the Feather RP2040. I've tried two separate CPBs and two different pots (same pot, one is on a breakout, one is off the breakout - this one here: https://www.adafruit.com/product/4493 and this one https://www.adafruit.com/product/4133).
[image]

tannewt, 2021-08-09:
Please try the absolute latest. I don't think this has been fixed but it's always good to check.

gallaugher, 2021-08-09:
@tannewt have been chatting with @danh on Discord & he pointed out that wiring to 3.3v on the CPB instead of CPX would be more appropriate. I just tried this & the "retreat" didn't occur - worked as expected. I was following the wiring diagram at: https://learn.adafruit.com/make-it-change-potentiometers/connections For the CPX I was using VOUT as well, though. I'm running 6.3 on all boards, so I think that's the latest release, unless you want me to try this out on 7.0. @danh mentioned the wiring diagram will be changed to advise power to 3.3 instead of VOUT. Wondering if I consider this a correction in the tutorial or if it should be addressed in CP, as well. As always, thx for your work!

dhalbert, 2021-08-09:
Closing due to fix to use proper voltage. I notified the Guide author.

closed 2021-08-09 by dhalbert