adafruit/circuitpython #5126: pre-commit doesn't fail when hook fails

microDev1, 2021-08-11:
With `uncrustify=0.69` (default with `Ubuntu 20.04`):

- when `python3 tools/codeformat.py -cv` is run, a bunch of errors can be seen
- when uncrustify is run with pre-commit, the errors go undetected and the formatting hook passes. 

My theory is that pre-commit expects the hook to return a certain error code on failure but `codeformat.py` isn't doing that.

jepler, 2021-08-11:
Keep in mind that uncrustify 0.69 is not supported. Adding a check that causes pre-commit to fail if that is the installed version would be welcome.

jepler, 2021-08-11:
As far as I know, this problem does not affect 0.71.

jepler, 2021-08-11:
We discussed this further on Discord, and think it would be a good idea to change codeformat to
 * add a specific version check
 * make a subprocess exiting with nonzero status cause codeformat to exit with nonzero status. Using `subprocess.check_call` is OK for this, even though it means it would stop processing files after the first error exit status.

PR #5137 (2021-08-13): Codeformat version checking
Add a version check for uncrustify.

Checking the return code doesn't work like we'd hoped, as a nonzero exit status is used by uncrustify to signify "success, but files modified".

(Because we disagree with some things uncrustify does, and un-do them by search & replace, it is normal for uncrustify to modify many many files every time it's invoked)

Closes: #5126
closed 2021-08-13 by tannewt