adafruit/circuitpython #5133: Display.refresh: no-args refresh() should never throw RuntimeError even if it takes a long time

jepler, 2021-08-12:
@todbot's [staroids](https://github.com/todbot/circuitpython_staroids) is awesome!  But as someone on Discord noted, it eventually crashes:
[image]

While I appreciate the intent behind the `minimum_frames_per_second` argument, my experience with it in practice has been that I always manually pass in `minimum_frames_per_second=0` or eventually experience a crash at the call to refresh.

I propose either to change the default to 0, or otherwise modify it so that the default `refresh()` call will not raise an exception (my other idea of using the return value doesn't seem to be workable). Most software in production mode doesn't want to crash entirely just because one frame took an unusually long time.

dhalbert, 2021-08-12:
Why is using the return value not workable? I agree the default should be zero, or it shouldn't be throwing an exception at all ever: the return value should tell you the status. Maybe return True, False, or None (not sure what the cases are, hence my question).

jepler, 2021-08-12:
If using the return value is workable, yay, that'd be great. Since True and False are given meanings already, would None indicate that the target frames per second was not reached, but a refresh occurred anyway?

jepler, 2021-08-12:
> When auto refresh is off, waits for the target frame rate and then refreshes the display, returning True. If the call has taken too long since the last refresh call for the given target frame rate, then the refresh returns False immediately without updating the screen to hopefully help getting caught up.

tannewt, 2021-08-12:
I'm ok changing the default minimum. I wouldn't change target since its meant to pace the code.

dhalbert, 2021-08-12:
I think this means:

- Too late per the target rate: returns False and do nothing
- Too late per the minimum rate: raises exception and does nothing
- Refresh happened: return True

I don't actually like encoding the return in True/False/None, that's probably a bad a idea. We could at least use an enum.

I wonder if we really need two targets. The best effort is to do the target. The program is struggling if it can't meet the minimum, but I'm not sure it should be an error it can't keep up.

> I'm ok changing the default minimum. I wouldn't change target since its meant to pace the code.

I agree, `0` would be a better default, and would mean that the minimum would be inactive most of the time, which is like just having one target. Then we don't need to fancy up the return value.

closed 2021-08-13 by microDev1

tannewt, 2021-08-13:
> I wonder if we really need two targets. The best effort is to do the target. The program is struggling if it can't meet the minimum, but I'm not sure it should be an error it can't keep up.

I added the minimum as a way to debug when the loop is way too slow.

PR #5138 (2021-08-13): Displayio: change refresh minimum_frames_per_second default to 0
Closes: #5133
commit e8df8297 (2021-08-13): Displayio: change refresh minimum_frames_per_second default to 0

Closes: #5133