adafruit/circuitpython #5134: stm32f411_blackpill_with_flash exposes pins dedicated to spi flash in board module

evildave666, 2021-08-12:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.6 on 2021-08-12; stm32f411ce-blackpill-with-flash with STM32F411CE
```

### Code/REPL

```python
import board
dir(board)
```

### Behavior

```python
['__class__', 'A0', 'A1', 'A10', 'A11', 'A12', 'A15', 'A2', 'A3', 'A4', 'A5', 'A6', 'A7', 'A8', 'A9', 'B0', 'B1', 'B10', 'B12', 'B13', 'B14', 'B15', 'B2', 'B3', 'B4', 'B5', 'B6', 'B7', 'B8', 'B9', 'C13', 'C14', 'C15', 'I2C', 'LED', 'SCL', 'SDA']

### Description

Pins A4-A7 are dedicated to the external spi flash chip which provides CIRCUITPY.

Per Scott, it is not correct behavior to expose these pins via the board module.

If I can make time I'll try to work on this one this weekend.

evildave666, 2021-08-12:
If I can make time I'll try to work on this over the weekend.

tannewt, 2021-08-24:
Didn't I change my mind on this because the pins are broken out on pin headers?

domdfcoding, 2022-01-06:
Does CircuitPython *need* to have exclusive use of the pins and the SPI bus for the flash chip, or could it share them with other SPI devices as long as they use a different CS pin? It seems a shame to lose use of the pins even when the flash isn't being read/written to.

tannewt, 2022-01-07:
> Does CircuitPython _need_ to have exclusive use of the pins and the SPI bus for the flash chip, or could it share them with other SPI devices as long as they use a different CS pin? It seems a shame to lose use of the pins even when the flash isn't being read/written to.

At this point yes. CircuitPython does have locking around a SPI bus but we haven't made sure it works ok to share a SPI bus between USB generated SPI flash accesses and user code accesses.

I think it'd be a lot of work to get right for little payoff. This is the only board out of 250+ that I know of with this issue.

domdfcoding, 2022-01-07:
Fair enough. I had suspected as much.

evildave666, 2022-02-21:
The flash is optional on this board (its basically a haxpress) and the pins are meant to be usable when the flash is not populated (the non-"with_flash" board). I still think that having board pins defined which probably don't work would cause more support issues than people wondering why they're not there if they're removed. The board does have multiple additional SPI ports available in any case.