adafruit/circuitpython #5136: Prevent clicks at the beginning or end of audio play

dhalbert, 2021-08-12:
Currently for some ports and some audio play mechanisms, there can be clicks at the beginning and/or the end of a sample due to sudden level shifts. We have support for quiescent values but these are often sample-file dependent.

Some ports ramp the amplitude up and down, but this prevents gapless playback and lengthens playing short samples (#3626).

We have also recommended in the past that users add ramping to their own audio files.

We should look at how this problem is solved in the real world, and try to emulate that. In many cases it may be done by speaker or amplifier disabling, which allows for a "soft" settling. Perhaps an audio enable/disable pin can be passed optionally to the audio play objects or methods.

makingsoundmachines, 2022-12-04:
We have been looking at the output from PWMAudio with a scope, and it seems that PWMAudioOut is adding a blip of audio unrelated to the actual sample into the first 4ms of the playback.

The blip is consistent when using the default buffer size and a buffer of bytearray(512), and changes to something different but consistent with other buffer sizes.

Here are some scope pictures where you can see what's happening:

The waveform in Audacity. The sample is mono, 44.1khz, 16 bit.

[image]

The waveform as output on RP2040, with the audio circuit as recommended in the RP2040 datasheet:

[image]

[image]

[image]

[image]

The relevant bits of code are

```
from audiocore import WaveFile
from audiopwmio import PWMAudioOut
..
audio_out1 = PWMAudioOut(board.GPIO24, right_channel=board.GPIO25 )
..
wave_file0 = open("/samples/808_A/BD_Hifi_1.wav", "rb")
wave0 = WaveFile(wave_file0, bytearray(512)) # same with and without bytearray(512)
wave0.sample_rate = 44100
..
while True:
    key_event = keys.events.get()
    if key_event:

            if key_event.key_number == 12:
              audio_out1.play(sine_wave_sample0, loop=True)
              time.sleep(0.1)
```

Hope this helps find the ploppy bug!

makingsoundmachines, 2022-12-10:
Update: using playback via the AudioMixer ( `mixer.voice[0].play()` ) is a workaround for now, as it only pops once when mixer.play is called and then all the consecutive plays do not pop.

So when using PWM Audio,
`audio_out1.play(mixer)` -> pops once at the beginning, but then `mixer.voice[0].play` doesn't pop
`audio_out1.play` pops everytime it's called

It's possible that the audio_dma_setup_playback called from common_hal_audiopwmio_pwmaudioout_play blocks the pwm being written long enough to cause the pop

andrewleech, 2023-02-17:
I've recently been playing with `audiomp3.MP3Decoder` and `audiopwmio.PWMAudioOut` on the rp2040 and find these clicks at the start and end of each track really distracting!

For reference I have a number of instrumental mp3 files encoded at 8kbps / 8Khz / mono. I've attached one here.
[Breathing In And Out.zip](https://github.com/adafruit/circuitpython/files/10763789/Breathing.In.And.Out.zip)
```
import os
import board
import audiomp3
import audiopwmio
import random
audio = audiopwmio.PWMAudioOut(board.GP0)
def random_shuffle(seq):
    l = len(seq)
    for i in range(l):
        j = random.randrange(l)
        seq[i], seq[j] = seq[j], seq[i]

    return seq

mp3files = random_shuffle(
    [f for f in os.listdir() if f.endswith(".mp3")]
)

mp3 = open(mp3files[0], "rb")
decoder = None

while True:
    for filename in mp3files:
        with open(filename, "rb") as mp3:
            if decoder is None:
                decoder = audiomp3.MP3Decoder(mp3)
            decoder.file = mp3
            audio.play(decoder)
            print("Playing:", filename)
            while audio.playing:
                time.sleep(0.1)

    random_shuffle(mp3files)

    if len(mp3files) > 1:
        while mp3files[0] == filename:
            random_shuffle(mp3files)
```
I get a click at before the first track, and again in between each track. I haven't tried the AudioMixer workaround mentioned above just yet...

I've been hunting through the C code for these modules and trying a bunch of things relating to buffers, decoding, chasing data from the mp3 files through decoder, dma and pwm trying to narrow down the source of the click. 

Finally stumbled on: `ports/raspberrypi/common-hal/audiopwmio/PWMAudioOut.c`
```
void common_hal_audiopwmio_pwmaudioout_stop(audiopwmio_pwmaudioout_obj_t *self) {
    if (self->pacing_timer < NUM_DMA_TIMERS) {
        dma_hw->timer[self->pacing_timer] = 0;
        self->pacing_timer = NUM_DMA_TIMERS;
    }

    audio_dma_stop(&self->dma);

    // // Set to quiescent level.
    printf("\nfinal: %d\n", common_hal_pwmio_pwmout_get_duty_cycle(&self->left_pwm));
    
    // common_hal_pwmio_pwmout_set_duty_cycle(&self->left_pwm, 255);
    // pwmio_pwmout_set_top(&self->left_pwm, PWM_TOP);
    if (self->stereo) {
        // common_hal_pwmio_pwmout_set_duty_cycle(&self->right_pwm, 255);
        // pwmio_pwmout_set_top(&self->right_pwm, PWM_TOP);
    }
}
```
I found that removing `common_hal_pwmio_pwmout_set_duty_cycle`  here removes the click in between each track! (not the one at turn on)
I've been playing with that funciton; I saw that at the end of the mp3 track linked get_duty_cycle() returns 255. Even enabling `set_duty_cycle(&self->left_pwm, 255);` to set the same value as returned brings the click back - so it seems that re-setting the pwm, even to the same value, triggers some sort of glitch in the output.

Interestingly though, removing the mp3 decoding and playing just a sine wave a few times over instead of mp3 doesn't cause any clicks. Not sure how this lines up with my above finding...
```
#Generate one period of sine wav.
length = 8000 // 880
sine_wave = array.array("H", [0] * length)
for i in range(length):
    sine_wave[i] = int(math.sin(math.pi * 2 * i / length) * (2 ** 15) + 2 ** 15)

sine_wave = audiocore.RawSample(sine_wave, sample_rate=8000)
audio.play(sine_wave, loop=True)
time.sleep(0.1)
audio.stop()
time.sleep(1)
audio.play(sine_wave, loop=True)
time.sleep(1)
audio.stop()
time.sleep(1)
```

KePere, 2024-05-16:
I've a similar problem.

We are using a i2S Adafruit microphone to acquire sound for a Raspi4B. We use arecord sentences according to the following link https://learn.adafruit.com/adafruit-i2s ... iring-test

We acquire audio signal every 3 seconds for instance, that is saves as a wavfile. The popping/clicking audio happens every start of the signal and sometimes also at the end.

I found some clues related with PWM of Raspi https://www.lightbluetouchpaper.org/201 ... pberry-pi/

There are also other similar post reporting the same instance

https://raspberrypi.stackexchange.com/q ... 985#147985

Can anyone help me with that?
Thnks