adafruit/circuitpython #5140: Support for Adafruit DYP ultrasonic Sonar Sensors

snehanyk05, 2021-08-12:
I am currently using a DYP-A01-V2 Ultrasonic sensor.
Datasheet in the link below:
[https://cdn-shop.adafruit.com/product-files/4665/4665_C14242.pdf](https://cdn-shop.adafruit.com/product-files/4665/4665_C14242.pdf)
I'm trying to find the right circuitpython library for it, but am unable to find one. Is there existing support for it or can i just go ahead and use the [HCSR04 adafruit](https://pypi.org/project/adafruit-circuitpython-hcsr04/) library istead?

ladyada, 2021-08-13:
there's no library - you can just use pyserial or UART serial
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/DYP_ultrasonics/me007ys.py

snehanyk05, 2021-08-13:
Got it to work! Thank you so much! :)

closed 2021-08-13 by snehanyk05