adafruit/circuitpython #5148: Tiny 2040

robdearpowa, 2021-08-13:
### CircuitPython version

```python
Adafruit CircuitPython 6.3.0 on 2021-06-01; Pimoroni Tiny 2040 with rp2040
```

### Code/REPL

```python
import busio
from board import *

print(dir(board))

i2c = busio.I2C(SCL, SDA)
print(i2c.scan())
i2c.deinit()
```

### Behavior

code.py output:
Traceback (chiamata più recente per ultima):
  File "code.py", riga 19, in <module>
NameError: nome 'SCL'non definito

Translation from italian:
  File "code.py", row 19, in <module>
NameError: name 'SCL' not defined

### Description

- board.I2C() doesn't provide a default I2C config
- board doesn't have SCL, SDA pins despite Tiny 2040 having them

tannewt, 2021-08-13:
@zodiusInfuser please take a look

ZodiusInfuser, 2021-08-13:
The board does not have specific default pins for I2C (or UART or SPI for that matter), as the pin numbering on the board is just 0-7 (versus the QTPy with explicit naming). The intention was to give users choice of which pins to use.

If defaults would be preferred so the above case works then I can make an executive decision on which ones, but would this prevent those pins from being used for other functions @tannewt?

robdearpowa, 2021-08-13:
If i may have a word, the real problem it's not that board.I2C() doesn't have a default pins for I2C, but that board itself doesn't provide a set of pins (example: SCL0, SDA0) to access them manually
[image]

Neradoc, 2021-08-13:
Since no pin are marked as default SDA or SCL on the sikkscreen or the pinout, it's perfectly normal not to have them in the `board` module. That's the case for the pico for example. However `board.I2C`, `board.SPI` and `board.UART` should be removed since they just raise `NotImplementedError`.

Default I2C pins do not impact what the users can do, `board.I2C` is only inited on first use, the names are aliases, a pin can be both named SDA and D5 for example.

@robdearpowa you probably want to do something like this for example (with whatever pins you are using)
```py
busio.I2C(sda=board.GP26, scl=board.GP27)
```

robdearpowa, 2021-08-13:
> Since no pin are marked as default SDA or SCL on the silkscreen or the pinout, it's perfectly normal not to have them in the `board` module. That's the case for the pico for example. However `board.I2C`, `board.SPI` and `board.UART` should be removed since they just raise `NotImplementedError`.
> 
> Default I2C pins do not impact what the users can do, `board.I2C` is only inited on first use, the names are aliases, a pin can be both named SDA and D5 for example.
> 
> @robdearpowa you probably want to do something like this for example (with whatever pins you are using)
> 
> ```python
> busio.I2C(sda=board.GP26, scl=board.GP27)
> ```
Ohh thanks a lot, i tought you could create buis.I2C() only via SCL, SDA namings, thanks

closed 2021-08-13 by dhalbert