adafruit/circuitpython #5150: macropad.pixels[0] flashes randomly+/- after "code done running"

w8HAQRHkTx7r, 2021-08-14:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.5 on 2021-07-21; Adafruit Macropad RP2040 with rp2040
```

### Code/REPL

```python
from adafruit_macropad import MacroPad
import time

macropad = MacroPad()

for i in range(10):
    time.sleep(0.25)
```

### Behavior

After the for loop completes, "code done running" is displayed and pixels[0] will begin to flash red randomly.   

### Description

The "code done running" is expected.  The flashing is not.  I would expect the default behavior to be "all pixels off".  If a
```
while True:
  pass
```
is added to the code, the code is never done and pixels[0] never flashes.

If pixels[0] is illuminated at the time the code falls thru the bottom, the pixel goes dark, but will begin to flash randomly.

### Additional information

Sometimes, the flashes are momentary (blink-of-an-eye), sometimes the light stays on for 4+/- seconds before going out.
Sometimes, the pixel will flash blue or green, but once flashing starts, the color remains the same -- at least until power is removed.
The flashing occurs whether the MacroPad is powered on the usb of my laptop, whether Mu Editor is running or not, or whether the MacroPad is powered by a dump usb charging port on my surge protector.
At one point, at the suggestion of a Forum admin, I added
```
macropad.display.auto_refresh = False
macropad.pixels.auto_write = False
```
which had no effect on the pixel. 

This happened on an early release of CircuitPython 7, and continues to fail on 20210806.

dhalbert, 2021-08-14:
The first NeoPixel is used as a status pixel, like the status NeoPixels on Feather and Metro boards. If your program starts using the NeoPixels, the status information will not be displayed. Normally your user program would run forever, but if it completes, CircuitPython restart and will again display status information.

w8HAQRHkTx7r, 2021-08-14:
@dhalbert I'm sorry I'm not familiar with the Feather nor Metro and I don't understand your comment.  Are you saying the random flashes are providing me with some sort of status information?

dhalbert, 2021-08-14:
Yes, the status RGB LED indicates various states and errors. In 6.x and earlier, it flashed more often: https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#circuitpython-rgb-status-light-2978455-21. In 7.0.0, we are simplifying it and reducing the frequency of flashing, to save power and annoyance. The new flashing scheme hasn't been written up yet in detail.

w8HAQRHkTx7r, 2021-08-15:
I guess that puts an end to it.

tannewt, 2021-08-16:
Could you post a video? It shouldn't be random. If the code completes without error it'll be a single green blink every 5 seconds or so. If an exception ends the code, it'll blink red twice every 5 seconds. In safe mode it is three yellow blinks.

w8HAQRHkTx7r, 2021-08-16:
Very interesting

w8HAQRHkTx7r, 2021-08-16:
While trying to video the tests, I ran into issues of consistent results.  My first test was 
```
from adafruit_macropad import MacroPad

macropad = MacroPad()
```
The p[0] flashed GREEN every 5 seconds, just like you implied.   This is the first time I'd seen that.

The next test was
```
from adafruit_macropad import MacroPad
import time

macropad = MacroPad()

for i in range(10):
    time.sleep(0.25)
```
and it double-flashed RED every 5 seconds.  

When I added a
```
macropad.pixels[i] = 0xffffff
```
into the body of the for loop and got single (not double)  red flashes, occasional solid red, a solid green.  This is what I've referred to as random sequence of flashes.

I went back to the original two-line test and get single flashes of red, not solid greens like I had originally.

I have 10 minutes of video.  I'll process it and post a link to it.

Neradoc, 2021-08-16:
I can repro, using the following code.py:
```py
from adafruit_macropad import MacroPad
import time
macropad = MacroPad()
for i in range(10):
    time.sleep(0.25)
    macropad.pixels[i] = 0xffffff
```
I got the same inconsistent red blinks. Then changed it to this and saved:
```py
import board
import neopixel
import time
pix = neopixel.NeoPixel(board.NEOPIXEL, 12)
pix.fill((255, 255, 255))
time.sleep(5)
```
An got this instead of all white.

[image]

dhalbert, 2021-08-16:
I do not see these problems, both with the tip of `main` and with alpha.5. I also updated the macropad library to the absolute latest and it's ok. If you haven't updated the library, try that.

I tried doing just red, green, and blue, and they look OK also.

dhalbert, 2021-08-16:
Try also putting a delay of a second or so at the beginning of the program. I have a weak hypothesis...

w8HAQRHkTx7r, 2021-08-16:
I'm sorry.  It's long.  I ran multiple tests with inconsistent results.  Here's what a saw:

https://youtu.be/CNROk2Ach_Y

Summary:
1) Normal green blinks
2) Normal double-red blinks
3) Random / single-red flash or 5 second solid
4) Ditto

w8HAQRHkTx7r, 2021-08-16:
Typical random behavior with

```
from adafruit_macropad import MacroPad
import time
time.sleep(3)
macropad = MacroPad()

for i in range(10):
    macropad.pixels[i] = 0xff0000
    time.sleep(0.25)
```

dhalbert, 2021-08-16:
What is your host operating system, and what editor are you using? Do you have any utility programs installed that are backup programs, disk utilities, etc.?

If you install a program such as above as code.py, and then plug the macropad into an AC adapter (so no host computer), do you see the same random flashing? If you press the reset button, does the behavior repeat exactly?

Neradoc, 2021-08-17:
I have it on latest, I even flash-nuke'd, put head from S3, and reinstalled the libraries from circup.
With just w8HAQRHkTx7r's code.py I get at minimum the red flash (after the code runs successfully).
Plugging the board into a power outlet with the same code.py I get the same result.

I noticed that the code running with the macropad library seems to have the correct colors though, so I looked at what else it initializes and I added that to my second file (the one without the macropad library that displays wrong) and it made the colors show correctly:
```py
import rotaryio
_encoder = rotaryio.IncrementalEncoder(board.ROTA, board.ROTB)
```

From that I've written a looping test and ran it on a power outlet with the same result:
- colors right in code.py
- colors wrong in lights.py as is
- colors right in lights.py if I add the rotary encoder setup

`code.py`
```py
import board
import neopixel
import time

import rotaryio
_encoder = rotaryio.IncrementalEncoder(board.ROTA, board.ROTB)

pix = neopixel.NeoPixel(board.NEOPIXEL, 12)
for i in range(12):
    time.sleep(0.25)
    pix[i] = (255, 255, 0)
    print("CODE RUNNING")

import supervisor
supervisor.set_next_code_file("/lights.py")
supervisor.reload()
```
`lights.py`
```py
import board
import neopixel
import time

pix = neopixel.NeoPixel(board.NEOPIXEL, 12)
for i in range(12):
    time.sleep(0.25)
    pix[i] = (0, 255, 255)
    print("LIGHTS RUNNING")

import supervisor
supervisor.reload()
```

dhalbert, 2021-08-17:
This might be a PIO issue: neopixels uses PIO, and rotary encoder also uses PIO.

w8HAQRHkTx7r, 2021-08-17:
I'm on the latest MacOS.  I don't have any backups running or other utilities to manage disks.

The problem happens on a raw AC outlet. I use Mu Editor. 

I'll try a Reset later and an alternate way of changing code.py.

w8HAQRHkTx7r, 2021-08-17:
I tried to be methodical and took good notes in this series of tests.  I created 4 .py files in Mu and saved them with the following names:
```
/Volumes/CIRCUITPY $ cat twoline.py 
print("twoline")
from adafruit_macropad import MacroPad
macropad = MacroPad()

/Volumes/CIRCUITPY $ cat importtime.py 
print("importtime")
from adafruit_macropad import MacroPad
import time
macropad = MacroPad()

/Volumes/CIRCUITPY $ cat sleeploop.py 
print("sleeploop")
from adafruit_macropad import MacroPad
import time
macropad = MacroPad()
for i in range(10):
    time.sleep(0.25)

/Volumes/CIRCUITPY $ cat lightpixels.py 
print("lightpixels")
from adafruit_macropad import MacroPad
import time
macropad = MacroPad()
for i in range(10):
    macropad.pixels[i] = 0xff0000
    time.sleep(0.25)
```

Then I exited Mu and shutdown my MacBook Pro.  After booting up, I plugged in the MacroPad to the usb port, opened a terminal window and switched from one program to another using:
```
cp <program> code.py
```
and recorded the status of pixels[0] when code done was reached.  Here are the results:

- twolines.py
- Random
- Reset button
- Green every 5
- importtime.py
- Green
- sleeploop.py
- Green
- lightpixels
- RANDOM
- Reset button
- RANDOM
- twolines.py
- RANDOM
- Rese button
- Green
- sleeploop.py
- Green

Interesting.  As I'd sensed earlier, once the MacroPad gets into its random funk, it tends to stay there -- even for .py files that would otherwise be green.  It takes something like a reset to clear the funk.

dhalbert, 2021-08-17:
@Neradoc could you try a `microcontroller.reset()` instead of the reload, to see if it's a hw setup issue? Thanks.

Neradoc, 2021-08-17:
It works correctly after a reset, with the button or `microcontroller.reset`, until `rotaryio.IncrementalEncoder` is deinited. I've done a few tests, it seems the issue appears after the `rotaryio.IncrementalEncoder` is deinited, but only if the neopixels were used before. Being connected to a computer or not has no incidence on any of that.

This displays all green as expected:
```py
pix = neopixel.NeoPixel(board.NEOPIXEL, 12)
encoder = rotaryio.IncrementalEncoder(board.ROTA, board.ROTB)
time.sleep(2)  # just in case
encoder.deinit()
pix.fill(0x00FF00)
time.sleep(2)
```
This displays all blue, then wrong colors:
```py
pix = neopixel.NeoPixel(board.NEOPIXEL, 12)
encoder = rotaryio.IncrementalEncoder(board.ROTA, board.ROTB)
pix.fill(0x0000FF)
time.sleep(2)
encoder.deinit()
pix.fill(0x00FF00)
time.sleep(2)
```
I don't know anything about PIO unfortunately, but it feels like deiniting the IncrementalEncoder leaves the PIO in some state, interfering with the memory or the pins of the other one. Like, it's reading the encoder and writing into the neopixels buffer maybe ? I'll leave that to the experts.

tannewt, 2021-08-17:
@w8HAQRHkTx7r thanks for the video and your patience. We'll get this fixed soon.

dhalbert, 2021-08-22:
Another report of sometimes-wrong NeoPixel colors, in the Arduino library (which I would guess uses the same PIO code):
https://github.com/adafruit/Adafruit_NeoPixel/issues/288