adafruit/circuitpython #5154: repeated modules (qrio,array)

jerryneedell, 2021-08-15:
@jepler Tagging at your request.

 I just did a build for esp32s2 and noticed that help('modules') shows qrio and array twice. 
It does not seem to cause any problems, but I assume it is not desired.
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.6-57-g498755af0 on 2021-08-15; Saola 1 w/Wrover with ESP32S2
>>> help('modules')
__main__          collections       micropython       ssl
_bleio            countio           msgpack           storage
_pixelbuf         digitalio         neopixel_write    struct
adafruit_bus_device                 displayio         onewireio         supervisor
adafruit_pixelbuf dualbank          os                synthio
alarm             errno             ps2io             sys
analogio          espidf            pulseio           terminalio
array             fontio            pwmio             time
array             framebufferio     qrio              touchio
audiobusio        frequencyio       qrio              traceback
audiocore         gc                rainbowio         ulab
audiomixer        getpass           random            usb_cdc
binascii          imagecapture      re                usb_hid
bitbangio         io                rgbmatrix         usb_midi
bitmaptools       ipaddress         rotaryio          vectorio
board             json              rtc               watchdog
builtins          keypad            sdcardio          wifi
busio             math              sharpdisplay
canio             microcontroller   socketpool
Plus any modules on the filesystem
>>>
```

jepler, 2021-08-16:
I believe this happens because `qrio` and `array` both happen to use `MP_REGISTER_MODULE` in addition to being listed "in some other location": `array` directly in `objmodule.c` and `qrio` via `MICROPY_PORT_BUILTIN_MODULES`.

I made `qrio` this way so that I could build it into the Unix port with a minimum of fuss. I actually think it may be the preferable way, but @tannewt I'd love your input about that.

tannewt, 2021-08-16:
I'm not sure what you are asking me.

jepler, 2021-08-16:
@tannewt how do you feel about using the `MP_REGISTER_MODULE` style to list shared-bindings python modules, going forward?

tannewt, 2021-08-17:
I haven't looked into it at all so I'd defer to you.

PR #5166 (2021-08-20): Don't double-list modules that MP_REGISTER_MODULE
Closes: #5154
commit ccc371ed (2021-08-20): Don't double-list modules that MP_REGISTER_MODULE

Closes: #5154
closed 2021-08-20 by microDev1