adafruit/circuitpython #5155: AttributeError: type object 'object' has no attribute '__new__' (Regression?)

nitz, 2021-08-15:
### CircuitPython version

```python
CircuitPython 7.0.0-alpha.6-dirty on 2021-08-14 (built from 82a952b6aa93eb1fcbf60c52584ac5c122fb3a56); Adafruit QT Py
```

### Code/REPL

```python
class Foo:
    _instance = None

    def __new__(cls):
        if not cls._instance:
            cls._instance = object.__new__(cls)
        return cls._instance

f = Foo()
```

### Behavior

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 10, in <module>
  File "code.py", line 6, in __new__
AttributeError: type object 'object' has no attribute '__new__'
```

### Description

Seems to be a regression of sorts. I found #1437, which describes the same behavior (using `super()` rather than `object`, though it happens with both: `AttributeError: 'super' object has no attribute '__new__'`). As shown in the code, this is a pure python class with no native base (using the inferred `object` base.)

I'm running a custom CP build for the QT Py (samd) with the following .mk options set:

```make
INTERNAL_FLASH_FILESYSTEM = 1
LONGINT_IMPL = NONE

# No room for a full build.
CIRCUITPY_FULL_BUILD = 0

# So, add some non-default full build options ...
CIRCUITPY_CONSOLE_UART = 1
CIRCUITPY_I2CPERIPHERAL = 1
CIRCUITPY_KEYPAD = 1

# ... but make some room via bits I don't need
CIRCUITPY_ANALOGIO = 0
CIRCUITPY_BUSIO_SPI = 0
CIRCUITPY_GETPASS = 0
CIRCUITPY_PWMIO = 0
CIRCUITPY_RANDOM = 0
CIRCUITPY_ROTARYIO = 0
CIRCUITPY_STRUCT = 0
CIRCUITPY_USB_MIDI = 0
```

Adding `MICROPY_CPYTHON_COMPAT = 1` results in an identical flash size remaining, so it seems that is still getting set. As well, it looks like that's [the only condition](https://github.com/adafruit/circuitpython/blob/main/py/objobject.c#L46) that needs to be set to have `object__new__(mp_obj_t cls)` build.

### Additional information

Please don't forget to have a lovely day ♥

DavePutz, 2021-08-15:
It looks like the issue is, indeed, MICROPY_CPYTHON_COMPAT. It is being set unconditionally in py/circuitpy_mpconfig.h to the value of CIRCUITPY_FULL_BUILD. So, setting MICROPY_CPYTHON_COMPAT = 1  in your .mk file won't do anything. I tested by setting it to 1 in py/circuitpy_mpconfig.h and your script runs OK.

nitz, 2021-08-16:
Fantastic find, Dave! I honestly hadn't even checked the header, only looking at the `.mk`. It looks like the header still uses the  `MICROPY_CPYTHON_COMPAT` symbol for at least a string compare bit in there, I wonder if the use of `CIRCUITPY_FULL_BUILD` for `MICROPY_CPYTHON_COMPAT` was an accident? I'm replying from mobile but I'll dig through blames when I make it back to my PC.

This definitely should work around my small issue in the mean time though!

dhalbert, 2021-08-16:
The setting is to save space in small builds.

nitz, 2021-08-16:
Sure, that makes plenty of sense @dhalbert, but is there a reason `MICROPY_CPYTHON_COMPAT` shouldn't be opt-in even when `CIRCUITPY_FULL_BUILD ` is 0? Like @DavePutz pointed out, it's [tied directly to `CIRCUITPY_FULL_BUILD`](https://github.com/adafruit/circuitpython/blob/498755af0180e699b72b7773c3ca5265a329f60c/py/circuitpy_mpconfig.h#L191) as it stands. now.

It looks like in d0a210654767677d6d26f1d4d40c378e03e9ca8e on April 22, 2020,, it was changed from allowing it to be set regardless of `CIRCUITPY_FULL_BUILD` to just directly using the value: [here](https://github.com/adafruit/circuitpython/commit/d0a210654767677d6d26f1d4d40c378e03e9ca8e#diff-14579858b50e6fe4e330bdd3a9d79d3477ae34d36262c7bd4bdff9a9227df603).

Switching the value to 1 for the board that I'm on right now results in a delta of 2168 bytes. I definitely understand the need to save space on these small flash controllers, but it feels like `MICROPY_CPYTHON_COMPAT` being unable to even be opted into on non-full builds makes no sense to me.

```
# with  `#define MICROPY_CPYTHON_COMPAT                (1)`
183184 bytes used, 4976 bytes free in flash firmware space out of 188160 bytes (183.75kB).
10296 bytes used, 22472 bytes free in ram for stack and heap out of 32768 bytes (32.0kB).

#with `#define MICROPY_CPYTHON_COMPAT                (CIRCUITPY_FULL_BUILD)`
181016 bytes used, 7144 bytes free in flash firmware space out of 188160 bytes (183.75kB).
10296 bytes used, 22472 bytes free in ram for stack and heap out of 32768 bytes (32.0kB).

dhalbert, 2021-08-16:
@nitz Yes, it should be `?=`, to allow optionally setting it, not just `=`. I think it was set unconditionally several years ago and we should have changed it. Perhaps we should re-purpose this issue to fix that.

tannewt, 2021-08-16:
Fixed by #5157.

closed 2021-08-16 by tannewt