adafruit/circuitpython #5159: FunHouse I2C Fails with External Sensor Attached

lcongdon, 2021-08-16:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.6 on 2021-08-12; Adafruit FunHouse with ESP32S2 (problem also occurs in version 6.3.0)
```

### Code/REPL

```python
Adafruit CircuitPython 7.0.0-alpha.6 on 2021-08-12; Adafruit FunHouse with ESP32S2
>>> import board
>>> i2c = board.I2C()
>>> import adafruit_dps310
>>> dps = adafruit_dps310.DPS310(i2c)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_dps310.py", line 225, in __init__
ValueError: No I2C device at address: 77
>>> import adafruit_pct2075
>>> pct = adafruit_pct2075.PCT2075(i2c)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_pct2075.py", line 103, in __init__
ValueError: No I2C device at address: 37
>>> i2c.try_lock()
True
>>> [f'0x{addr:02X}' for addr in i2c.scan()]
['0x37', '0x38', '0x77']
```

### Behavior

I2C fails on FunHouse if an external I2C sensor is attached.

With no external I2C sensor attached, onboard sensor functions as expected:
```
Adafruit CircuitPython 7.0.0-alpha.6 on 2021-08-12; Adafruit FunHouse with ESP32S2
>>> import board
>>> i2c = board.I2C()
>>> import adafruit_dps310
>>> dps = adafruit_dps310.DPS310(i2c)
>>> dps.temperature
40.5015
```
Attaching an external sensor causes I2C to fail for both onboard and external sensors.

### Description

FunHouse functions as expected with no external I2C device attached. Adding an external sensor causes the I2C bus to fail.

### Additional information

Error occurs with different sensors, different cables, and in two versions of CircuitPython.

At times, the I2C scan works as shown in the example. At times, it hangs.

This problem also occurs when using the FunHouse layered software. That is, the fragment:
```
Adafruit CircuitPython 7.0.0-alpha.6 on 2021-08-12; Adafruit FunHouse with ESP32S2
>>> from adafruit_funhouse import FunHouse
>>> funhouse = FunHouse()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_funhouse/__init__.py", line 104, in __init__
  File "adafruit_funhouse/peripherals.py", line 81, in __init__
  File "adafruit_dps310.py", line 225, in __init__
ValueError: No I2C device at address: 77
```
fails when an external sensor is attached via I2C.

tannewt, 2021-08-16:
What sensors specifically do you have issue with? How long are the wires you are using to connect them?

lcongdon, 2021-08-16:
The problem occurs with the PCT2075, the SGP40, and the BHT1750. Each fail with either 2 inch or 4 inch cables. I have reverted to CircuitPython 6.3.0 due to other issues with 7.0.0 alpha and the issue is present in 6.3 as well.

skerr92, 2021-08-17:
I did want to see if this was an issue specific to Funhouse vs a broader issue with ESP32-S2. I tried this with the ESP32-S2 PixelWing I just had added and tested with 7.0 Alpha with two i2c boards. One the MPU-6050 and the other the 1.3" OLED display.

I wasn't able to replicate the issues with my ESP32-S2 board so it's possible it could be isolated to Funhouse.

dgriswo, 2021-08-18:
I have a FunHouse and tried with a PCT2075, BH1750, SGP30 (all concurrent) with CircuitPython 6.3.0.  I was able to instantiate an object for all the sensors and the FunHouse object without issue.  I repeated the test under 7.0.0-alpha.6 without issue. I don't have an SGP40 to test.

I even tried pulling A0 on the PCT2075 high to make its i2c address 0x77 and couldn't reproduce the "No I2C device at address: 77" error. (It didn't work, but didn't generate the same error)

dhalbert, 2021-08-18:
@lcongdon Let's see if this is a possible hardware problem. Thanks.

- Are you using the Adafruit sensor boards, or some other boards?
-  Which STEMMA cables are you using? Are you using Adafruit ones? Could you give the part numbers your ordered?
- Are you using the small black STEMMA/QT socket? Are you using cables directly to the boards, or are you going to header pins?
- Could you measure the voltage at the terminals on a sensor board, checking Vin, SCL, SDA, and ground?

KingerNorth, 2021-08-18:
I tested the Funhouse and added 2 breakouts with stemmaqt connectors, the apds9960 and the vl6180. The i2c and the Funhouse house worked perfectly. I tried it with both adafruit-circuitpython-adafruit_funhouse-en_US-7.0.0-alpha.6.uf2 and the latest file adafruit-circuitpython-adafruit_funhouse-en_US-20210817-5b0009c.uf2. I was using 100m stemma cables between the Funhouse and both breakouts. Also the on board aht20 was funning during all of the test. All okay for me.

lcongdon, 2021-08-18:
@dhalbert, your question prompted a close inspection of the I2C socket on the FunHouse and there is a bent pin. Thanks for looking into this.

closed 2021-08-18 by lcongdon

dhalbert, 2021-08-18:
> @dhalbert, your question prompted a close inspection of the I2C socket on the FunHouse and there is a bent pin. Thanks for looking into this.

Glad it was easy to spot. Thanks for checking. Sometime's it just mechanical!

lcongdon, 2021-08-18:
@skerr92, @dgriswo and @KingerNorth thanks also for checking. I incorrectly assumed a software problem. Working for me now as well.