adafruit/circuitpython #5169: IL0373 Epaper - quick refresh

bmeisels, 2021-08-17:
Hi,

@amit794 and I have been working on quick refresh support (by using a different waveform) on the IL0373 as part of our work on the ARAMCON 2 Badge.
Our code can be found at (friendly warning - this is not production grade):
CircuitPython core changes - https://github.com/aramcon-badge/circuitpython/tree/feature/refresh-modes
IL0373 driver - https://github.com/aramcon-badge/aramcon-firmware/blob/master/lib/display.py.

We would like to merge our changes upstream (of course with a redesign to fit better into CP).
The question from our side is would quick refresh support be accepted into CP core (and the Adafruit IL0373 driver)?
I understand if this would be problematic due to the risk of destroying user hardware but I believe this risk can be mitigated by giving a known good configuration and forcing full-refresh after a specific amount of quick refresh cycles.

tannewt, 2021-08-18:
Please make a draft PR for the changes to the core. That makes it easier to see what you are proposing for the core.

I think we'd be open to merging it in but would want to know the details first.

bmeisels, 2021-08-28:
Now that #5199 has been merged we can close this.
Thanks!

closed 2021-08-28 by bmeisels