adafruit/circuitpython #5176: Circuit Python Core Crash

sa-schaefer, 2021-08-18:
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Attempted heap allocation when VM not running.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.4 on 2021-07-08; Adafruit Feather RP2040 with rp2040

dhalbert, 2021-08-18:
Please provide more details ("Please file an issue with the contents of your CIRCUITPY drive"), and if this is reproducible, try alpha.6 or "Absolute Newest".

sa-schaefer, 2021-08-18:
Not sure what files to attach as there are files that I am not sharing.  I have a copy of the CIRCUITPY drive I can attach required files to.
I have to go, will look at this later

closed 2021-08-18 by sa-schaefer

sa-schaefer, 2021-08-18:
I am back, I rebooted the Feather RP2040 and the problem went away.  Not sure what happened, if it reoccurs I will try the alpha 6 or Absolute Newest.