adafruit/circuitpython #5183: Convert all circuitpython modules to use MP_REGISTER_MODULE

jepler, 2021-08-19:
This can reduce the amount of boilerplate required.  Instead of introducing a macro such as `QRIO_MODULE` dependent on `CIRCUITPY_QRIO` and then adding an expansion of it in `MICROPY_PORT_BUILTIN_MODULES_STRONG_LINKS`, one simply writes in the bindings `__init__.c` file:
```
MP_REGISTER_MODULE(MP_QSTR_qrio, qrio_module, CIRCUITPY_QRIO);
```
where the first item is the module name as a qstr, the second item is the module's globals, and the third is the preprocessor macro that expands to (1) when enabled.

This can be done on a piecemail basis, and this bug would be closed once the last one is completed. I didn't prepare a checklist here, but basically everything in shared-bindings should be done unless a reason not to is identified.

Besides reducing boilerplate slightly, it also eases including modules with no hardware dependency in the "unix" build, which enables them to be testable. In time, such testability could be applied to bitmaptools and atexit, to name just two examples.

capellini, 2021-08-24:
@jepler I'd like to help with this.  I've been experimenting with this using displayio and my pybadge.

capellini, 2021-09-01:
I think that I've now converted all modules in shared bindings that have a `*_MODULE` defined in `py/circuitpy_mpconfig.h`.  I didn't see a module listed in `MICROPY_PORT_BUILTIN_MODULES_STRONG_LINKS` for the following modules in shared bindings, so didn't use `MP_REGISTER_MODULE`:

- multiterminal
- paralleldisplay
- nvm (submodule of microcontroller)
- os (listed in `MICROPY_PORT_BUILTIN_MODULE_WEAK_LINKS` and alt name listed in `MICROPY_PORT_BUILTIN_MODULE_ALT_NAMES` and I don't quite know how to deal with those yet)
- time (same reasons as 'os' above)

There are seven more modules defined in `MICROPY_PORT_BUILTIN_MODULES_STRONG_LINKS`:

- `BINASCII_MODULE`
- `ERRNO_MODULE`
- `ESPIDF_MODULE`
- `JSON_MODULE`
- `RE_MODULE`
- `RP2PIO_MODULE`
- `SAMD_MODULE`
 
I started to register these with `MP_REGISTER_MODULE` but then thought it best if I consult with you before doing so, as I'm new to circuitpython and micropython.  Even if I should proceed with these, I figured I should keep them in a separate PR, since dealing with them may be more nuanced than the shared-bindings boilerplate code.  Would you like me to give a PR for these seven a shot?

jepler, 2021-09-01:
@capellini I'm "out of the office" so I can't look into it in detail right now. If you want to try to PR just one of those changes, perhaps someone else can review it and give useful feedback. 

@tannewt do you know why those modules are different, or if we need to handle the situation differently?

tannewt, 2021-09-01:
`espidf`, `rp2pio` and `samd` are port specific modules so I'd do those as a group.

The other four are micropython modules that we reorganized into shared-bindings. `os` and `time` are in this group too. I think they can be made strong links. MicroPython has `uos` and `utime` too which is why they have strong and weak links.

I'm not sure `multiterminal` is used anywhere. We used to use it on ESP8266 for webrepl but I'm not sure we'd go that route when we support ESP32 again.

`paralleldisplay` is brand new and should follow our others.

commit 0eafb672 (2021-10-07): Convert all modules to use MP_REGISTER_MODULE

Convert binascii, errno, json, and re to use MP_REGISTER_MODULE.

Resolves #5183.