adafruit/circuitpython #5186: Print function doesn't print the GPIO state correctly.

waiweng83, 2021-08-20:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.6-103-gf9f106b58 on 2021-08-20; Cytron Maker Pi RP2040 with rp2040
```

### Code/REPL

```python
import board
import digitalio
import time

button = digitalio.DigitalInOut(board.GP20)
button.direction = digitalio.Direction.INPUT

led = digitalio.DigitalInOut(board.GP0)
led.direction = digitalio.Direction.OUTPUT

while True:
    led.value = not button.value
    print(button.value)
    
    #time.sleep(0.1)
```

### Behavior

The function print(button.value) doesn't print out the GPIO state correctly if I commented out the time.sleep(0.1).
It works fine if the 0.1s delay is there.

In both cases, the LED does react correctly to the button's state.

Neradoc, 2021-08-20:
I don't see you set a pull direction in your code ?
```py
button = digitalio.DigitalInOut(board.GP20)
button.direction = digitalio.Direction.INPUT
button.pull = digitalio.Pull.UP
```
Alternatively you can use this:
```py
button = digitalio.DigitalInOut(board.GP20)
button.switch_to_input(digitalio.Pull.UP)
```

waiweng83, 2021-08-20:
> I don't see you set a pull direction in your code ?
> Does the board have a pull up ?
> 
> ```python
> button = digitalio.DigitalInOut(board.GP20)
> button.direction = digitalio.Direction.INPUT
> button.pull = digitalio.Pull.UP
> ```
> 
> Alternatively you can use this:
> 
> ```python
> button = digitalio.DigitalInOut(board.GP20)
> button.switch_to_input(digitalio.Pull.UP)
> ```

It doesn't matter as there is external pull up on my board (Maker Pi RP2040).
Also the LED does behave expectedly according to the state of the push button.
Only the print function doesn't print the actual value when there is no delay in the loop.

dhalbert, 2021-08-20:
I think you may be seeing switch bounce. You are reading the button state twice, and the second time may be different than the first, due to bouncing. The delay does debouncing for you.

Also, the LED value is the opposite of the button state. You did not say what you expected, just that you thought it was wrong.

Try this:
```py
while True:
    button_now = button.value
    led.value = not button_now
    print(button_now)
    
    #time.sleep(0.1)
```

waiweng83, 2021-08-20:
> I think you may be seeing switch bounce. You are reading the button state twice, and the second time may be different than the first, due to bouncing. The delay does debouncing for you.
> 
> Also, the LED value is the opposite of the button state. You did not say what you expected, just that you thought it was wrong.
> 
> Try this:
> 
> ```python
> while True:
>     button_now = button.value
>     led.value = not button_now
>     print(button_now)
>     
>     #time.sleep(0.1)
> ```

I'm sorry that if I didn't make it clear.

I have a button which is active low and a LED which is active high.
I expect that when I press the button, the LED will turn on and the printed value on REPL will be False.

However, when I run the code, the LED does turn on when I press the button, but the output to the REPL remains as True.
I've tried your code and the problem is still there.

anecdata, 2021-08-20:
I don't have the Maker board, but I was unable to replicate this on an QT Py RP2040, mirroring the Cytron Maker Pi RP2040 button circuit here: https://drive.google.com/file/d/1Zp8GYO8x7ThObB1G8RIZx2YdqrXtdUc0/view
The LED and print() both reflected the correct state, regardless of delay.

waiweng83, 2021-08-20:
> I don't have the Maker board, but I was unable to replicate this on an QT Py RP2040 running absolute newest from S3 (`7.0.0-alpha.6-121-ga98363348 on 2021-08-20`), mirroring the Cytron Maker Pi RP2040 button circuit here: https://drive.google.com/file/d/1Zp8GYO8x7ThObB1G8RIZx2YdqrXtdUc0/view
> The LED and print() both reflected the correct state, regardless of delay.

This is strange. I'm able to reproduce the problem running QT Py RP2040 firmware on Maker Pi RP2040.
I'm also able to reproduce this on Raspberry Pi Pico.

dhalbert, 2021-08-20:
Suppose you remove the LED part of the code and just print the value of the button over and over. Does it go False when you press it and True when you release? Do you see the same thing with the other button, on GP21?

waiweng83, 2021-08-23:
> Suppose you remove the LED part of the code and just print the value of the button over and over. Does it go False when you press it and True when you release? Do you see the same thing with the other button, on GP21?

The problem still persists after removing the LED part. It also happens on other GPIO.

waiweng83, 2021-08-23:
I think I've found out the root cause for this problem.
I've done some test using the code below:

```python
import board
import digitalio
import time

led = digitalio.DigitalInOut(board.GP0)
led.direction = digitalio.Direction.OUTPUT

count = 0

while True:
    if count > 10000:
        led.value = 1
    print(count)
    count += 1
```

And what I got was the LED turns on way before the printed counter is > 10000.

Can I confirm that the print function is non-blocking and calling the print function is actually just adding the value to a queue and it will only be printed after the previous value. This way if we keep calling the print function faster than it runs, the queue grow bigger and bigger and the actual printed value will lag way behind.

dhalbert, 2021-08-23:
> Can I confirm that the print function is non-blocking and calling the print function is actually just adding the value to a queue and it will only be printed after the previous value.

There's a buffer for USB output, and the output is also buffered on the host side. On the host side, the buffer can get very large, so you'll have some lag. When and if the host-side buffer fills up, the USB driver there will do a USB stall to prevent getting more input, and the CircuitPython side will not transmit more information until the stall is released. Some terminal programs are faster than others. Mu is Python based (if that's what you're using), and may be slower at keeping up. What terminal program are you using?

> I'm not sure whether it's intended to be so, but is there a way to make sure the printed value is actually the latest value (besides putting a delay in between loop)?

There's no way to tell when the output actually appears in your terminal program. All we know is when it got transmitted to the host.

Instead of printing the button value over and over, the best thing to do is to print its value only when it changes. For this, you can use the `adafruit_debouncer` library (https://learn.adafruit.com/debouncer-library-python-circuitpython-buttons-sensors) or the new `keypad` built-in module in CircuitPython 7 (https://learn.adafruit.com/key-pad-matrix-scanning-in-circuitpython). Or you can just keep track of the current value of the button, and print it only when it changes.

Normally, you'd use the debouncer library or `keypad` anyway to debounce a bare switch, since all mechanical buttons bounce (see the first link above). The Cytron board actually does hardware debouncing with a capacitor, but that's unusual.

waiweng83, 2021-08-24:
@dhalbert Thanks for your explanation. I understand it clearly now.
So it seems like it's not a bug and I will close this issue.

Thanks everyone.

closed 2021-08-24 by waiweng83