adafruit/circuitpython #5187: Add aliases for I2S audio pins to board module

NathanY3G, 2021-08-20:
In the `board` module there are aliases for pins that have certain functions e.g. `SD_CS`. It would be convenient if similar aliases existed for the I2S audio pins. This would make it unnecessary to add conditional logic similar to the following:

```
board_name = os.uname().machine

if "Grand Central M4 Express" in board_name:
    i2s = audiobusio.I2SOut(board.D14, board.D33, board.D32)
elif # etc.
```

I believe a number of boards have fixed I2S pin assignments including:

1. Adafruit Feather M4 CAN Express
2. Adafruit Feather STM32F405 Express
3. Adafruit Grand Central M4 Express

dhalbert, 2021-08-20:
The functionally labelled pins usually correspond to either silkscreen markings or hardwired connections (e.g. `NEOPIXEL`, `SD-CS`). We don't tend to assign function names to pins such as the Grand Central pins above when you'd have to look up the pins anyway. Though there are boards where only certain pins can be used for certain functions, like the I2S examples above, there are other boards where the choice of pins is much wider. For instance, the nRF chips can do I2S on any pins.

The SAMD51 actually supports I2C on two sets of pins -- it's just that some of them are not connected on some boards.

So for these reasons we wouldn't do this. But it would perhaps make sense on some boards to label the silkscreen functionally, if there is room (which unfortunately there often is not).

There are related discussions about default pins on the Pi Pico, which overlaps somewhat with this one: #4121 and #5041, though reasons for not doing it on the Pico are not exactly the same as this.

closed 2021-08-20 by dhalbert

NathanY3G, 2021-08-21:
Thanks @dhalbert for taking the time to explain the rationale. I believe I understand the intent of the `board` module now.