adafruit/circuitpython #5192: Translations with two-byte unicode characters shouldn't use two bytes for every character

tannewt, 2021-08-20:
When any character that is compressed is two bytes, then all characters in the `values` and `words` tables will be twobytes. Instead, only the mulibyte character should take multiple bytes.

[image]

This change caused the french version to increase ~500 bytes simply due to two fancy apostrophe's in one phrase.

cwalther, 2021-08-20:
In other words, the compression should work in terms of UTF-8 bytes, not in terms of codepoints (characters).

Docteh, 2021-08-21:
I think the compression needs to see which method is more efficient. Most languages using code points with high values are probably using lots of code points with high values. 

Throwing a few fancy apostrophes into the middle of many latin characters seems to have been unexpected.

The fancy apostrophe in question is 8217 in hex `0x2019` 
Also represented as three bytes in UTF-8
```
echo -n ’ |hexdump -C
00000000  e2 80 99                                          |...|
00000003
```

https://github.com/adafruit/circuitpython/blob/1825fc86d3316747040148445d3ec81f0f765487/locale/ja.po#L143-L145
The string is 21 characters long, which would be 42 bytes, but represented in UTF-8 this particular string is 53 bytes
```
echo -n "%qは長さ2のタプルでなければなりません" | hexdump -C
00000000  25 71 e3 81 af e9 95 b7  e3 81 95 32 e3 81 ae e3  |%q.........2....|
00000010  82 bf e3 83 97 e3 83 ab  e3 81 a7 e3 81 aa e3 81  |................|
00000020  91 e3 82 8c e3 81 b0 e3  81 aa e3 82 8a e3 81 be  |................|
00000030  e3 81 9b e3 82 93                                 |......|
00000036
```

tannewt, 2021-08-21:
FYI: I discussed this on my stream today: https://www.youtube.com/watch?v=veOmVVP_YsY

jepler, 2021-08-21:
The portion that you saw showing up as two bytes per character is the "dictionary", which is a small portion of the message data overall. (called "words" in `genhdr/compression.generated.h`)

It turns out this is not a savings for all languages, basically for the reasons @Docteh mentions above, we need make the compression system accommodate either situation and determine which is more efficient. (Another alternative would be to use an variable length encoding that indexes into the array called `values`; `values` is again in a constant width encoding and its values are either character values or pointers back into the dictionary of "words".  This would give a good distribution of mostly 1-byte encoded values with a few 2-byte encoded values, for any language)

That said, I prototyped the unconditional use of UTF-8 so we could get the statistics. Here is a selection of languages on trinket_m0:

|translation|before|after|delta|
|--|--|--|--|
|fr| 1068| 1448| +380|
|de| 1308| 1316|   +8|
|zh| 1580| 1744| +164|
|ja| 1380|  872| -508|

jepler, 2021-08-23:
I had an additional idea, which was to make the entries in "words" index "values" rather than being code points; the indexing is "utf8-like" if more than 0-255 indices in values are needed.  This should be a win on average, as (no matter the language) most code points in "words" should be represented by just 1 byte. However, it ends up a bit worse for latin languages still.

| language | before | after | delta |
|--|--|--|--|
| fr | 1328 | 1312 | -16 |
| de_DE | 1052 | 1032  | -20 |
| ja | 1124 | 1284 | +160 |
| zh_Latn_pinyin | 1324 | 1564 | +240 |

My code: https://github.com/jepler/circuitpython/pull/new/wordsutf8

Because we need to index "values" I don't think it can be stored as utf-8, but that doesn't mean there's not a better way to store it.

jepler, 2021-08-24:
I doodled around with ideas to reduce the size of the huffman table in the ja translation, but I don't think there ends up being much gained, or maybe even a net loss.

General idea: Split code points into 8:8 values. Find out the number of distinct upper octets (70 in the case of the ja translation today), and store items as a combination of the lower 8 bits plus an index for the upper bits. In this case, though, only 1 bit per huffman table is saved encoding in this way, but you have to pay for the decoding function (about 40 bytes estimated) and the 70 bytes for the upper-bits table. As there are only about 600 entries in the table, nothing is gained.

dhalbert, 2022-02-14:
The Russian translation is now the largest one, even though it's perhaps only 20-25% done. I think this may be due to this issue.

dhalbert, 2023-10-28:
@jepler is this addressed in some way now?

jepler, 2023-10-29:
We now error at build time if a language should have been 8 bits but isn't

closed 2023-10-29 by jepler