adafruit/circuitpython #5198: Macropad RP2040 sometimes doesn't report position change.

PeterFalken, 2021-08-21:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-alpha.6 on 2021-08-12; Adafruit Macropad RP2040 with rp2040
```

### Code/REPL

```python
## Imports
from adafruit_macropad import MacroPad

## Global variables
os_index = 0
app_index = 0

l_position = 0  ## last position

## Init
macropad = MacroPad()
text_lines = macropad.display_text(title="MacroPad Info")

## Main loop
while True:
    diff = macropad.encoder - l_position
    l_position = macropad.encoder
    if 0 != diff:
        if macropad.encoder_switch:
            if diff > 0:
                os_index+=1
            else:
                os_index-=1
            ## Reset app_selector
            app_index = 0
        else:
            if diff > 0:
                app_index+=1
            else:
                app_index-=1

    text_lines[0].text = "OS Selector {}".format(os_index % 3)
    text_lines[1].text = "App Selector {}".format(app_index % 12)
    text_lines[2].text = "Encoder switch: {}".format(macropad.encoder_switch)
    text_lines.show()
```

### Behavior

No error message on console.

### Description

If turning the rotary encoder fast - sometimes the app_index variable won't change value.
I'm trying to update the Macropad Hotkeys code so that pushing the encoder switch will change to different folder containing specific OS macros (basically grouping hotkey macro definitions by OS; Linux, macOS, Windows, etc.)

The onboard encoder has 20 stops per full rotation - each rotation should add/decrease the app_index variable in 20 units. But sometimes the increment is 10 units or 14 units.

### Additional information

Connected through USB to macOS, but his issue should be OS independent.

dhalbert, 2021-08-21:
IIn the code above, trying getting the `macropad.encoder` value only once, since it may vary between the two calls at the beginning of the loop.

The rotary encoder handling is interrupt-driven, and sometimes can lose clicks when the encoder transitions happen too fast. I'm not sure that's the issue here. There is also a known problem about reversing: #3875. We've spent time tweaking the code, but there's a tradeoff between accuracy in terms of registering false transitions vs losing transitions.

PeterFalken, 2021-08-21:
Thanks Dan.
I'll update the code so that macropad.encoder is only evaluated once and then run a few tests.
`## Main loop
while True:
    n_position = macropad.encoder
    diff = n_position - l_position
    l_position = n_position
    if 0 != diff:
...`

I'll report back with my findings.

PeterFalken, 2021-08-22:
After updating the code to only call _macropad.encoder_ - the behavior is improved, less positions are lost if turning the encoder fast. I think that in the end the positional code should actually be something like this:

```python
while True:
    n_position = macropad.encoder
    diff = n_position - l_position
    l_position = n_position
    if 0 != diff:
        if macropad.encoder_switch:
            os_index+=diff
            ## Reset app_selector
            app_index = 0
        else:
            app_index+=diff
        ## Switch App Hotkeys & exit loop.
        apps[app_index % len(apps)].switch()
        continue
```

dhalbert, 2021-08-22:
Glad it's working better. Are you modifying this code? https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/Macropad_Hotkeys/code.py
If you think it could use improvement, feel free to submit a PR, though a feature addition would require a guide update, so I'm not sure we'd do that right now.

Do you think we should close this issue?

PeterFalken, 2021-08-22:
Dan,
Let's close this issue, since the root cause is already being tracked on the ticket you mentioned on the first comment.
I'll submit a PR for the Macropad Hotkeys with some improvements. The feature addition - I'll work on it and get in touch once it's ready for release, I could probably take care of updating the guide as well.

Thanks for the help, I learned a couple of things.

closed 2021-08-22 by PeterFalken