adafruit/circuitpython #5201: Raytac nrf52840 Dongle not booting

jerryneedell, 2021-08-22:
I tried using one of the new RAYTAC-nrf52840 dongles? https://www.adafruit.com/product/5199 With the pre-installed firmware, I can see it with the nrf-connect app and see changes in the button state and well as set the LED on/off. I tried uploading CP from tip of main to  it and it does not seem to boot - I see several errors in dmesg related to the file system. Does this device need some special handling?
dmesg errors 
```
[439587.025403]  sdd:
[439587.027901] sd 8:0:0:0: [sdd] Attached SCSI removable disk
[439607.466351] usb 3-3.2: USB disconnect, device number 8
[439607.481272] blk_update_request: I/O error, dev sdd, sector 11 op 0x1:(WRITE) flags 0x0 phys_seg 8 prio class 0
[439607.481307] Buffer I/O error on dev sdd, logical block 11, lost async page write
[439607.481311] Buffer I/O error on dev sdd, logical block 12, lost async page write
[439607.481313] Buffer I/O error on dev sdd, logical block 13, lost async page write
[439607.481315] Buffer I/O error on dev sdd, logical block 14, lost async page write
[439607.481317] Buffer I/O error on dev sdd, logical block 15, lost async page write
[439607.481327] Buffer I/O error on dev sdd, logical block 16, lost async page write
[439607.481329] Buffer I/O error on dev sdd, logical block 17, lost async page write
[439607.481330] Buffer I/O error on dev sdd, logical block 18, lost async page write
[439607.481344] blk_update_request: I/O error, dev sdd, sector 269 op 0x1:(WRITE) flags 0x0 phys_seg 8 prio class 0
[439607.481347] Buffer I/O error on dev sdd, logical block 269, lost async page write
[439607.481349] Buffer I/O error on dev sdd, logical block 270, lost async page write
[439607.481379] blk_update_request: I/O error, dev sdd, sector 517 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
```

dhalbert, 2021-08-22:
I tried this the dongle with:
- https://adafruit-circuit-python.s3.amazonaws.com/bin/raytac_mdbt50q-rx/en_US/adafruit-circuitpython-raytac_mdbt50q-rx-en_US-7.0.0-alpha.6.uf2: works and shows CIRCUITPY
- https://adafruit-circuit-python.s3.amazonaws.com/bin/raytac_mdbt50q-rx/en_US/adafruit-circuitpython-raytac_mdbt50q-rx-en_US-20210822-fdfcaf0.uf2: does not work

So a bisect is in order.

dhalbert, 2021-08-22:
I tried a Feather nRF52840 with fdcaf0 as well, and it doesn't work either. So something about the nRF builds has broken since alpha.6.

dhalbert, 2021-08-22:
Bisected to
- d2860b58b06f02303a30a7ff505f3fae2c21ba25, which is #5182 

[The title of an issue or PR will be included in the rendering, _if_ you put it in a list.]

PR #5202 (2021-08-23): nrf: remove critical section around sd_app_evt_wait()
- Fixes #5201.

#5182 added a critical section around `sd_app_evt_wait()`, which prevents USB from coming up (and perhaps other things).
https://github.com/adafruit/circuitpython/blob/fdfcaf0b72c61165922c63985909fef0b4868260/ports/nrf/supervisor/port.c#L367-L372
This removes that critical section, which was not present before.

Other uses I see of `sd_app_evt_wait()` don't do anything special (no critical sections, etc.)

@tannewt I'm not sure if this the most correct fix, based on what your intention was, which was to shut off as many inte [... cut]
closed 2021-08-23 by tannewt