adafruit/circuitpython #5216: Enable key velocity for MIDI keybeds by adding "ticks_ms" property to keypad.Event

todbot, 2021-08-24:
The `keypad` module can be used for MIDI keyboards, and provide velocity sensitivity, with the addition of a millisecond timestamp to the `keypad.Event` object.

In a recent [@jedgarpark  livestream](https://www.youtube.com/watch?v=bVuzj_N_MKc), he demo'd a MIDI keyboard using `keypad`. Like most keybeds, the one demo'd is implemented as two matrices: one for key down, one for key up. The intended use is key velocity, computed by measuring the time between "keypress begin" (key-up matrix 'release') and "keypress end" (key-down matrix 'press'). Millisecond resolution is sufficient for this.

In use, the user's code maps a `key_number` to "keypress begin" or "keypress end". On "keypress end" event, use timestamp of matching previous "keypress begin" to compute velocity.

The benefit of this approach is that it's a minimal addition to `keypad` and does not impact normal users of `keypad`.  The downside of this approach is that, for users wanting velocity, it requires some key state management and dealing with the somewhat confusing "key press begin" event being a "release" event. But I figure this could be wrapped up in a helper library to minimize those downsides.

This is a very low-priority feature request.

dhalbert, 2021-08-25:
I have had other requests for a timestamp for keypad events for other use cases, and it would be relatively easy to implement this.

dhalbert, 2023-10-28:
Fixed by #5360.

closed 2023-10-28 by dhalbert