adafruit/circuitpython #5254: am trying to strobe several PWM GPIO and play a sound at the same time and get:

dhalbert, 2021-08-27:
am trying to strobe several PWM GPIO and play a sound at the same time and get:
RuntimeError: All timers in use
if I use less then 10 GPIOs I can get the sound, more and the error occurs. If I can put the sound onto another process I think i can bypass the timer issue??? thoughts?

_Originally posted by @LarryPriest in https://github.com/adafruit/circuitpython/issues/4542#issuecomment-907488603_

dhalbert, 2021-08-27:
There are limited number of hardware timers on any particular chip. So multiprocessing will not help you: you have exhausted the hardware resources. What board is this?

LarryPriest, 2021-08-28:
pico, i am working on several sub-modules of a K-9 project. i have the memory map ( bigclivedotcom gallium) and a binary counter but when I tried to add sound i ran out of timers. oh well, next step -> . . . thanks for the insight.

dhalbert, 2021-08-28:
You could play audio with `audiobusio.I2SOut`, but you'll need an I2S amplifier (we have those). You could try that as a substitute for `AudioPWMOut` now even if you can't hear what you and see if you run out of anything.

If you need lots of PWM's, you can get a breakout board that uses a PCM9685, which has up to 16 channels on some breakouts. You control it with I2C. See https://www.adafruit.com/product/815 and https://www.adafruit.com/product/2928, for example. There are others.

I will close this issue now because it's now really project help. Our discord server (https://adafru.it/discord) or the forums (https://forums.adafruit.com) are the best places to ask for further help on thiis.

closed 2021-08-28 by dhalbert

LarryPriest, 2021-08-28:
thanks again -ttyl