adafruit/circuitpython #5258: esp32s2 - imagecapture sometimes adds extra byte

jerryneedell, 2021-08-28:
### CircuitPython version

```python
dafruit CircuitPython 7.0.0-beta.0-110-gb90a16b67 on 2021-08-28; Saola 1 w/Wrover with ESP32S2
```

### Code/REPL

```python
This is a modified version of the ov2640 library example. The changes print out the first/lat 10 bytes of each image and if the extra byte is present, it is stripped when the image is encoded.

 board
import busio
import digitalio
import wifi
import socketpool
import adafruit_minimqtt.adafruit_minimqtt as MQTT
from adafruit_io.adafruit_io import IO_MQTT
import adafruit_ov2640

feed_name = "image-saola-ov2640"

print("Connecting to WIFI")
wifi.radio.connect(secrets["ssid"], secrets["password"])
pool = socketpool.SocketPool(wifi.radio)

print("Connecting to Adafruit IO")
mqtt_client = MQTT.MQTT(
    broker="io.adafruit.com",
    username=secrets["aio_username"],
    password=secrets["aio_key"],
    socket_pool=pool,
    ssl_context=ssl.create_default_context(),
)
mqtt_client.connect()
io = IO_MQTT(mqtt_client)

with digitalio.DigitalInOut(board.IO10) as reset:
    reset.switch_to_output(False)
    time.sleep(0.001)

bus = busio.I2C(scl=board.IO7, sda=board.IO8)
cam = adafruit_ov2640.OV2640(
    bus,
    data_pins=(
        board.IO36,
        board.IO37,
        board.IO41,
        board.IO42,
        board.IO39,
        board.IO40,
        board.IO21,
        board.IO38,
    ),
    clock=board.IO33,
    vsync=board.IO2,
    href=board.IO3,
    mclk=board.IO1,
    mclk_frequency=20_000_000,
    size=adafruit_ov2640.OV2640_SIZE_QVGA,
    shutdown=board.IO14,
    reset=board.IO12,
)

cam.flip_x = False
cam.flip_y = False

cam.size = adafruit_ov2640.OV2640_SIZE_SVGA
cam.colorspace = adafruit_ov2640.OV2640_COLOR_JPEG
jpeg_buffer = bytearray(cam.capture_buffer_size)
while True:
    jpeg = cam.capture(jpeg_buffer)
    print(f"Captured {len(jpeg)} bytes of jpeg data")
    print([hex(x) for x in jpeg[0:10]])
    print([hex(x) for x in jpeg[len(jpeg)-10:len(jpeg)]])
    # b2a_base64() appends a trailing newline, which IO does not like
    if jpeg[0] == 0xff and jpeg[1] == 0xff:
        encoded_data = binascii.b2a_base64(jpeg[1:]).strip()
    else:
        encoded_data = binascii.b2a_base64(jpeg).strip()
    print(f"Expanded to {len(encoded_data)} for IO upload")
    print(encoded_data[0:10])
    io.publish(feed_name, encoded_data)
    print("Waiting 10s")
    time.sleep(10)
```

### Behavior

 unpredictably the image capture will be normal (starting with 0xff 0xd8) and other times there will be an extra 0xff byte at the beginning of the image buffer. If not removed, the encoded image will not be recognizable by AIO.

rebooting or resetting will often change from one behavior to another. It also rarely happens spontaneously whlie the code is exdecuting.

I have only seen this on the Saola -- I have not seen it on the Kaluga...yet..

edited REPL output -- only showing the first/last 10 bytes of the capture buffer

The first example shows an extra 0xff byte pre-pended to the image data
```
Captured 71752 bytes of jpeg data
['0xff', '0xff', '0xd8', '0xff', '0xe0', '0x0', '0x10', '0x4a', '0x46', '0x49']
['0xea', '0x5f', '0xe', '0xf', '0x4a', '0x75', '0x68', '0x41', '0xff', '0xd9']
Waiting 10s
Captured 71421 bytes of jpeg data
['0xff', '0xff', '0xd8', '0xff', '0xe0', '0x0', '0x10', '0x4a', '0x46', '0x49']
['0xfa', '0xca', '0x5f', '0xe6', '0xac', '0x8a', '0xb5', '0xcf', '0xff', '0xd9']

```
after a reboot(or two - it varies) the image is correctly presented without the extra byte.
```
Press any key to enter the REPL. Use CTRL-D to reload.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Connecting to WIFI
Connecting to Adafruit IO
Captured 75345 bytes of jpeg data
['0xff', '0xd8', '0xff', '0xe0', '0x0', '0x10', '0x4a', '0x46', '0x49', '0x46']
['0xbf', '0xdc', '0xad', '0x14', '0x74', '0xb9', '0xe', '0x47', '0xff', '0xd9']
Waiting 10s
Captured 70354 bytes of jpeg data
['0xff', '0xd8', '0xff', '0xe0', '0x0', '0x10', '0x4a', '0x46', '0x49', '0x46']
['0x4d', '0xdd', '0x6c', '0xae', '0x25', '0xef', '0x3b', '0x1f', '0xff', '0xd9']

```

dhalbert, 2021-12-16:
@jepler Any idea about this?

jerryneedell, 2021-12-16:
FYI -- I updated CP and libraries and retried -- now only the first capture after reboot is valid -- all subsequent images are not able to be decoded by AIO

```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.2.0-alpha.0-146-g190aee551 on 2021-12-16; Saola 1 w/Wrover with ESP32S2
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Connecting to WIFI
Connecting to Adafruit IO
Captured 35401 bytes of jpeg data
['0xff', '0xd8', '0xff', '0xe0', '0x0', '0x10', '0x4a', '0x46', '0x49', '0x46']
['0x77', '0xa', '0xaa', '0x14', '0xb0', '0x19', '0xa0', '0xf', '0xff', '0xd9']
Expanded to 47204 for IO upload
b'/9j/4AAQSk'
Waiting 10s
Captured 6862 bytes of jpeg data
['0x68', '0x6d', '0x17', '0x62', '0xb9', '0xf7', '0xab', '0x91', '0xcf', '0x4e']
['0x5e', '0xe1', '0x55', '0xb6', '0x93', '0xd6', '0x80', '0x3f', '0xff', '0xd9']
Expanded to 9152 for IO upload
b'aG0XYrn3q5'
Waiting 10s
```

jepler, 2021-12-16:
I haven't tried to investigate or fix this since Jerry initially reported it.

jepler, 2022-07-06:
Feel free to open a fresh bug if this issue is still happening.