adafruit/circuitpython #5259: Option to disable waiting for safe mode

deshipu, 2021-08-28:
I was reading comments on a bug, and stumbled upon this comment:
https://github.com/adafruit/circuitpython/issues/5212#issuecomment-907307341

which explains why my keyboards are so slow to work when I'm switching between hosts with a KVM. I previously blamed the slow USB enumeration in general, and the KVM in particular, but this explains it.

I wonder if adding a compile-time option for skipping this wait (so that I could use it for my keyboards) would be accepted? In my case I don't particularly need the safe mode, and this would increase the comfort of working with a KVM greatly.

I also wonder if there are more such delays that could be disabled?

tannewt, 2021-08-30:
Compile time option is fine with me. I wouldn't have it on by default though.

nRF will have a second wait time for the BLE workflow.

Perhaps we could add a safe mode pin to check at some point or maybe even a key matrix. Those checks would be fast but still allow for safe mode.

deshipu, 2021-09-02:
Fixed by #5270

closed 2021-09-02 by deshipu