adafruit/circuitpython #5266: audiobusio.I2SOut: audio distorted "crackly" playing MP3

nathanielbutts, 2021-08-30:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-beta.0/Raspberry Pico
```

### Code/REPL

```python
import board
import audiocore
import audiomp3
import audiobusio
import digitalio
import time

f = open("/music/15-03.mp3", "rb")
mp3 = audiomp3.MP3Decoder(f)
a = audiobusio.I2SOut(board.GP27, board.GP28, board.GP26)

print("playing")
a.play(mp3)
while a.playing:
  time.sleep(.001)
print("stopped")
```

### Behavior

I'm still having this issue on Pico RP2040, using 7.0.0-beta.0. There is a difference in the crackling between beta.0 and alpha.6. On alpha.6 it was the same crackling for the entire length of play back. On beta.0 it cycles, kinda like:
`_|¯|_|¯|_|¯|______|¯|_|¯|_|¯|______|¯|_|¯|_|¯|_`

### Description

Issue presents itself using both:

- [Waveshare Pico-Audio Hat](https://www.waveshare.com/wiki/Pico-Audio)
- [Adafruit MAX98357 I2S Amp](https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp)

I am able to use the same hardware and play WAV files without displaying the same condition.

### Additional information

[Link to original issue and discussions from @PaintYourDragon](https://github.com/adafruit/circuitpython/issues/5047#issuecomment-907825956)

tannewt, 2021-08-30:
What is the bitrate of the MP3? You may need to lower it.

kattni, 2021-08-30:
These are the MP3 restraints for RP2040/Pico. Supported MP3 files are mono and stereo files less than 64kbit/s, with sample rates from 8kHz to 24kHz. (This info was gathered for an upcoming guide, and I thought it might be helpful here.)

nathanielbutts, 2021-09-01:
I believe that was the problem.  Inspecting the file showed it to be 320kbit/s and 44.1khz, so basically .wav like.  I used xcfa to move it to 64kbit/s and the pops are gone.  It sounds kind of muted and scratchy, but that has probably to do more with manipulating the file too much.  I'll play around with ripping at the correct rates and try again.  

Thanks for looking at it!

closed 2021-09-01 by nathanielbutts