adafruit/circuitpython #5269: stm32: Meowbit board.MENU button can't be used — pin in use error

deshipu, 2021-08-30:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-beta.0-123-g89fc1890a-dirty on 2021-08-30; MEOWBIT with STM32F401xE
```

### Code/REPL

```python
>>> import board
>>> import digitalio
>>> digitalio.DigitalInOut(board.MENU)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: MENU in use
```

### Behavior

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: MENU in use
```

### Additional information

The ``board.MENU`` seems to be ``PC15``. I couldn't find any references to that pin in ``board.c``,  so I guess something is wrong with the code that checks whether the pin is in use or not?

dhalbert, 2021-09-09:
https://github.com/adafruit/circuitpython/blob/0ee0ed70350f029c0ebf05ceaeb89d5611ffa297/ports/stm/peripherals/stm32f4/stm32f401xe/gpio.c#L41
says `PC15` is `OSC32_OUT`, and so reserves it. There is no public schematic for the Meowbit, so I'm not sure if `OSC32` is used or not, and would even be available after boot.

deshipu, 2021-09-10:
Perhaps the pin for the menu button is wrong then.

Looking at https://www.kittenbot.cc/collections/frontpage/products/meowbit-codable-console-for-microsoft-makecode-arcade that button also enables DFU mode. I'm looking at the datasheet for STM32F401RET6 and the bootloader docs, but I can't see any reference to what pin that might be.

The micropython fork at https://github.com/KittenBot/micropython_meowbit simply doesn't define a menu pin at all.

Perhaps @kittenbot could comment on this?

deshipu, 2021-09-10:
Looking at the PCB traces under a microscope, it looks like the MENU button is indeed connected to pin PC15.