adafruit/circuitpython #5283: Sinewave tone playback on RP2040 no longer working as of CP7

kattni, 2021-09-01:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-beta.0-183-gb80ffee10 on 2021-09-01; Raspberry Pi Pico with rp2040
(but also the 7.0.0-beta.0 release)
```

### Code/REPL

```python
import time
import array
import math
import audiocore
import board
import audiobusio

I2S_CLK = board.GP0
I2S_WS = board.GP1
I2S_DATA = board.GP2

audio = audiobusio.I2SOut(I2S_CLK, I2S_WS, I2S_DATA)

tone_volume = 0.1  # Increase this to increase the volume of the tone.
frequency = 440  # Set this to the Hz of the tone you want to generate.
length = 8000 // frequency
sine_wave = array.array("h", [0] * length)
for i in range(length):
    sine_wave[i] = int((math.sin(math.pi * 2 * i / length)) * tone_volume * (2 ** 15 -1))
sine_wave_sample = audiocore.RawSample(sine_wave)

while True:
    audio.play(sine_wave_sample, loop=True)
    time.sleep(1)
    audio.stop()
    time.sleep(1)
```

### Behavior

The code attempts to transmit audio, but instead of a tone, a 1 second bit of semi-static-silence is played through the headphones/speaker, every 1 seconds.

Increasing the `tone_volume` variable does not resolve the issue.

### Description

Tested using the UDA1334A breakout with Raspberry Pi Pico and Feather RP2040. Same results with the MAX98357A breakout with a speaker connected on Raspberry Pi PIco.

This issue is not present in CP 6.3.0 on the Feather RP2040. 

Another tester was able to get tone playback on 6.3.0 on the Raspberry Pi Pico with the MAX98357A. I was unable to get tone playback on the Pico on CP6 or CP7 with the UDA1334A.

### Additional information

Possibly worth noting:
When testing on the Feather, I received the following error initially with my first attempt at wiring. 
```
ValueError: Bit clock and word select must be sequential pins
```
Further, MP3 playback works fine.

kattni, 2021-09-01:
So, update. WAV file playback is also resulting in silence on Pico running both 6.3.0 and 7.0.0-beta.0-183-gb80ffee10 using the UDA1334A. 

I now suspect the issue may lie with `audiocore`.

Please let me know if you need any further information, or have builds to test.

rsbohn, 2021-09-01:
Testing Pi Pico with UDA1334A.
With CP6 I hear the tone. Adafruit CircuitPython 6.3.0 on 2021-06-01; Raspberry Pi Pico with rp2040
With CP7 I hear static and some clicking. Adafruit CircuitPython 7.0.0-alpha.6-72-g162954a1a on 2021-08-17; Raspberry Pi Pico with rp2040
The static seems to be between the tones, or silence on CP7.

rsbohn, 2021-09-01:
I can play 44100 sample/sec .wav files on CP7.

dhalbert, 2021-09-01:
If you have an analog amp (TPA2012, etc., or try on the MacroPad), does `PWMAudioOut` work?

It appears there are conflicting test results of `RawSample` vs `WaveFile`. 

I had trouble with `I2SOut` due to miswiring sometimes.

I did `WaveFile` testing with the test program in #5196, adapted to the SAMD or RP2040 board I was testing on (`I2SOut` vs `PWMAudioOut`, and whether or not there was SPEAKER_ENABLE).

kattni, 2021-09-01:
@dhalbert I just did MP3 playback using PWM with the PAM8302 on the Pico and MacroPad - and tone playback definitely works on the MacroPad, it's built into the library.

Should I try something else?

SAMD51 worked with I2S and tone playback.

I'm concerned about my Pico because Carter and rsbohn are getting similar results to each other, but my results are odd.

dhalbert, 2021-09-01:
Triple-check the wiring is all I could suggest. The I2S amp I was using did not have the pins in the same order as the order on the board. I was testing with a Feather RP2040 because it has a reset button.

I think I was using the MAX98357A I2S amp.

dhalbert, 2021-09-01:
LRC is WS, BCLK is CLK, DIN is DATA, I think that's what I did on MAX98357A.

kattni, 2021-09-01:
Correct on the different order of pins from CircuitPython to the breakout - the UDA is clearer on pin names. I triple checked my wiring, and it appears to be right. Remember, it works with MP3 playback over I2S.

I ordered the MAX, I don't have it yet, I won't have it until Monday.

dhalbert, 2021-09-06:
The UDA1334A breakout cannot drive a speaker: it is line-level output meant to drive an amp. I tried a UDA1334A into a MAX98357A and I am hearing output. Are you testing the UDA1334A directly to a speaker, or with an amp in between.

I am hearing more scratchiness on `I2SOut` than `PWMAudioOut`, but that is a different issue.

dhalbert, 2021-09-06:
[deleted and reposted - previous post had several errors]

The UDA1334A breakout cannot drive a speaker: it is line-level output meant to drive an amp. I tried a UDA1334A into a TPA2012 analog amp, playing WAV files and I am hearing output. Are you testing the UDA1334A directly to a speaker, or with an amp in between?

I am not hearing the sine wave test at the top, though; I'm working on that.

BTW, I am hearing more scratchiness on I2SOut than PWMAudioOut, but that is a different issue.

dhalbert, 2021-09-07:
I have a fix that restores I2S output, by not trying to reuse the buffer supplied by `RawSample` as an output buffer. However, it is intermittently buzzy, which can sometimes only be fixed by a power-cycle. I hear this same buzziness in 6.3.0, so I think that was an existing problem.

The current RP2040 code does not do anything special to expand mono samples to stereo. I2S output is always stereo. By contrast, the SAMD implementation sets a special "mono" mode of the SAMD peripheral, which duplicates the input sample, and the ESP32-S2 implementation contains code that duplicates a sample into both channels. I think that we need to add mono->stereo support, which might be done in the sample conversion code, or could be done in the PIO code. I'm not sure why it works at all currently. But the DMA and the PIO output are running independently and it's possible they are getting out of sync.

kattni, 2021-09-07:
@dhalbert Here are the MP3s from my example.
[single-code.txt](https://github.com/adafruit/circuitpython/files/7123742/single-code.txt)

[mp3files.zip](https://github.com/adafruit/circuitpython/files/7123736/mp3files.zip)

tannewt, 2021-09-07:
FYI, the RP2040 memory system will expand narrow writes to 32 bits. This is how the 16 bit mono sample becomes a 32 bit stereo sample, the RP2040 mirrors the 16bits to the second position. See [the pico example](https://github.com/raspberrypi/pico-examples/blob/2fb08a028f886624ebc9c938775e054860549159/system/narrow_io_write/narrow_io_write.c) for details.

PR #5310 (2021-09-07): Fix RP2040 I2S: always copy to output buffer
- Fixes #5283.

Make RP2040 I2S work again.

- Don't attempt to use the sample buffer as the DMA buffer if no conversion is needed. Instead, always copy the sample buffer into the output buffer.
- Move the I2S PIO restart slightly to avoid starting before an error check.
- Add explanatory comment about getting mono->stereo conversion for free.

The quality of the I2S audio is not great: there are buzzes and noise, which happen erratically. A power cycle or reset can improve things temporarily. However, the quality seems to be the same as in CircuitP [... cut]
closed 2021-09-07 by kattni