adafruit/circuitpython #5305: Presence of boot.py in many cases will cause a boot loop on ESP32-S2 with TinyUF2 (rc, not beta)

anecdata, 2021-09-06:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-rc.1 on 2021-09-02; Adafruit FunHouse with ESP32S2
```

### Code/REPL

```python
None
```

### Behavior

On an ESP32-S2 device (Fun House in this case) without TinyUF2, device behaves as expected.

On the same device _with_ TinyUF2, starting clean from a fresh `storage.erase_filesystem()`, with either empty `boot.py` + empty `code.py`, or some (but not all) combinations of trivial and non-trivial `boot.py` + `code.py`, the device will restart and display the Blinka icon, then go into an apparent loop where nothing else is displayed (regardless of any print statements in `boot.py`) and the violet status lights and the red LED will flash every second.

To recover, go into safe mode and delete or rename the `boot.py` file, then hard reset.

Saw similar behaviors on `rc.0`, but not on the `beta`s.

anecdata, 2021-09-07:
I think it's more complicated... if I start clean, empty boot & code will work repeatedly. It seems like there's some unexpected state being retained in some cases, and once the error triggers, it will keep triggering across hard resets until boot.py is removed. It would almost have to be something in flash?

Another behavior I hadn't seen before: Some non-trivial `boot.py` and non-trivial `code.py` combinations will not trigger the looping, but the `code.py` will not run. `boot.py output:` will display correctly on the built-in display, and `boot-out.txt will be properly populated, but no `code.py output:` line or code output, simply `Code done running`.

tannewt, 2021-09-07:
Can you replicate this with a DEBUG build when it is outputting to the debug UART?

anecdata, 2021-09-08:
I haven't done a DEBUG build since it became too big to generate, but I'll ask around to see how to make it fit.

dhalbert, 2021-09-08:
> I haven't done a DEBUG build since it became too big to generate, but I'll ask around to see how to make it fit.

Paulsk cut the wifi buffers in half and made it fit: https://discord.com/channels/327254708534116352/537365702651150357/881908023436640356

anecdata, 2021-09-08:
Switched to Cucumber RIS since Funhouse doesn't have debug pins exposed.

1. erased flash
2. installed TinyUF2 combined
3. installed CP `rc.1`

Initial test with no `boot.py` and default `code.py` gives this in the debug console on startup:
```
# ...
I (904) cpu_start: Pro cpu up.
I (1318) spiram: SPI SRAM memory test OK
```

Creating an empty `boot.py` gave an ambiguous result on hard reset, repeatedly. No "Hello World!, but it's possible that `tio` didn't reconnect in time. It usually does. But this is the equivalent debug console out in this case:
```
# ...
I (1315) ??f̆?f?fφ~
                  ????
                      i?U?
                          R?V?KW????сz-m??j5R?ESP-ROM:esp32s2-rc4-20191025
```
The gibberish varies each time. Seems like something is getting clobbered.

anecdata, 2021-09-08:
If I put a larger `code.py` on it, it will go into a boot loop, but the debug console in this particular case looks more normal on each iteration (no gibberish):
```
# ...
I (915) cpu_start: Pro cpu up.
I (1329) spiram: SPI SRAM memory test OK
ESP-ROM:esp32s2-rc4-20191025
```

anecdata, 2021-09-08:
Manual bisect shows earliest S3 artifact with this issue is:
`2021-08-27T22:59:12.000Z 2.6 MB adafruit-circuitpython-gravitech_cucumber_rs-en_US-20210827-31b9dd4.uf2`
If I read the tea leaves right, I think that correlates to:
https://github.com/adafruit/circuitpython/pull/5245

anecdata, 2021-09-08:
Backing out just the additions on line 133 and 142 from that PR does indeed seem to fix the `boot.py` issue and allow `rc.1` to run successfully.

tannewt, 2021-09-09:
What are the contents of your boot.py and code.py? This does sound like the issue we thought we had fixed.

anecdata, 2021-09-09:
I'm trying but having difficulty coming up with a minimal reproducible example for the initial `code.py`.

@neradoc replicated the boot looping on FunHouse with an empty `boot.py` file and no `code.py`, but I don't know what his prior conditions were <https://discord.com/channels/327254708534116352/327298996332658690/884511468844879883> 

Necessary but insufficient conditions seem to be:
- Tiny UF2 (0.5.1 or 0.5.2)
- presence of `boot.py` file (can be empty)
- CircuitPython builds from `-en_US-20210827-31b9dd4.uf2` to present

Bu as far as I can tell, it needs those things but also has to be triggered by a `code.py` with presently unknown characteristics. Once it is triggered, the issue will persist through:
- successive hard resets
- going into bootloader mode
- going into safe mode

Once triggered, you can even go into safe mode and reduce `code.py` down to empty, and the issue will persist. After that point, only two things will fix it:

1. the deletion or renaming of `boot.py` and another hard reset will get things working again
2. resetting while at the "Press any key to enter the REPL. Use CTRL-D to reload." prompt and NOT from within the REPL, even if there is still a boot.py

I'll keep trying to get an example for an initial `code.py`.

anecdata, 2021-09-10:
I don't really understand what that PR does, or the issues that it addressed. Any ideas what _kinds_ of code might trigger this... large code size or complex code to interpret? big buffers? fragmentation? stack vs heap vs pystack vs esp-idf? other? I finally bodged together 600+ lines of code snippets that will consistently trigger the issue, but it's ugly and introduces  variables because a lot of modules get exercised. With some direction, I might be able to produce a more targeted example.

tannewt, 2021-09-10:
The issue we were trying to fix was a rare case where the ATB reads into the FTB. The ATB tracks block status and the FTB tracks whether each block has a finalizer. If the first block of the FTB looks like a TAIL block from the ATB, then the collector will read past the end of RAM and hang the chip. Once hung, the WDT will reset the chip after a time.

So it is a specific case of allocating something at the top of the heap and allocating something at the bottom with a finalizer. I talk about this in more detail in this stream: https://www.youtube.com/watch?v=DQVXNNUGvbk

I think the next step would be to print out the structure of the heap [here](https://github.com/adafruit/circuitpython/blob/main/py/gc.c#L191-L196) and you can validate the addresses read [here](https://github.com/adafruit/circuitpython/blob/main/py/gc.c#L248). The addresses (ptrs) should always be less than `0x3ff80000`. Reading that exact value hangs.

anecdata, 2021-09-10:
That's all way over my head. I missed that deep dive, so I'll watch and see what I can learn.

tannewt, 2021-09-13:
I wasn't able to replicate this on an UM FeatherS2. Have you only replicated it on devices with a screen?

anecdata, 2021-09-14:
It affects any ESP32-S2 device I've tried so far: MagTag, FunHouse, Cucumber RS (no display). If I delete some code, doesn't seem to matter which code, more like how much. Just verified that the latest build:
`adafruit-circuitpython-gravitech_cucumber_rs-en_US-20210913-3e9cf8f.uf2` still triggers the issue with the 600+ line conglomeration of snippets. I'll put it in a gist and link below.

FeatherS2 seems harder to trigger, but I don't know why. It has more PSRAM, but it's not like anything I've done fills the 2MB of PSRAM in the other boards.

dhalbert, 2021-09-14:
@anecdata I think you mentioned on Sunday that TinyUF2 is not necessary, at least with the example above? So let's edit the title, etc.

anecdata, 2021-09-14:
Just a note that in the large example, there are functions and function code that are never executed, but if I delete too much of that, the issue goes away. That leads me to think it has more to do with code size or complexity, and the interpretation to byte code stage. And indeed, if I have a `code.py` that simply imports the original offending but mpy-cross'd `{code}.mpy`, the issue does not arise. It seems to be the specific combination of a `boot.py`, plus a `code.py` actually containing certain (unknown) characteristics.

tannewt, 2021-09-14:
At what point does it fail with that `code.py`? It seems to get pretty far for me.

I also see you are enabling the WDT yourself. Please post more of your debug output. The S2 has multiple watchdogs and it should tell us which of the two reset the chip for you.

anecdata, 2021-09-14:
I don't think the user watchdog is required, it was just a handy snippet. The code will run and soft-reload fine. What it won't do is recover from a hard reset after it's done a `while True` or two. That results in boot loops with debug console like this:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x8 (TG1WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40028200
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x16a8
load:0x4004c000,len:0xeec
load:0x40050000,len:0x2bb8
entry 0x4004c310
I (24) boot: ESP-IDF v4.3-356-g48ae2309fd 2nd stage bootloader
I (24) boot: compile time 06:13:05
I (24) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s2: SPI Speed      : 80MHz
I (37) boot.esp32s2: SPI Mode       : QIO
I (42) boot.esp32s2: SPI Flash Size : 4MB
W (47) boot.esp32s2: PRO CPU has been reset by WDT.
W (52) boot.esp32s2: WDT reset info: PRO CPU PC=0x4004c671
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (82) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (90) boot:  2 ota_0            OTA app          00 10 00010000 00160000
I (97) boot:  3 ota_1            OTA app          00 11 00170000 00160000
I (105) boot:  4 uf2              factory app      00 00 002d0000 00040000
I (112) boot:  5 ffat             Unknown data     01 81 00310000 000f0000
I (120) boot: End of partition table
I (624) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=37c98h (228504) map
I (663) esp_image: segment 1: paddr=00047cc0 vaddr=3ffcd930 size=035fch ( 13820) load
I (666) esp_image: segment 2: paddr=0004b2c4 vaddr=40028000 size=04d54h ( 19796) load
I (674) esp_image: segment 3: paddr=00050020 vaddr=40080020 size=f0298h (983704) map
I (846) esp_image: segment 4: paddr=001402c0 vaddr=4002cd54 size=10bd4h ( 68564) load
I (862) esp_image: segment 5: paddr=00150e9c vaddr=40070000 size=0002ch (    44) load
I (862) esp_image: segment 6: paddr=00150ed0 vaddr=50000000 size=01010h (  4112) load
I (880) boot: Loaded app from partition at offset 0x10000
I (880) boot: Disabling RNG early entropy source...
I (893) cache: Instruction cache 	: size 16KB, 4Ways, cache line size 32Byte
I (893) cache: Data cache 		: size 16KB, 4Ways, cache line size 32Byte
I (898) spiram: Found 16MBit SPI RAM device
I (903) spiram: SPI RAM mode: sram 40m
I (908) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (915) cpu_start: Pro cpu up.
I (1329) spiram: SPI SRAM memory test OK
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x8 (TG1WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40083fcb
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x16a8
load:0x4004c000,len:0xeec
load:0x40050000,len:0x2bb8
entry 0x4004c310
```

PR #5348 (2021-09-15): Clear the buffer byte between ATB and FTB
This ensures it has a known start state. An unknown state risks it
looking like a tail block similar to the problem fixed by #5245.

Fixes #5305
commit e544909b (2021-09-15): Clear the buffer byte between ATB and FTB

This ensures it has a known start state. An unknown state risks it
looking like a tail block similar to the problem fixed by #5245.

Fixes #5305
closed 2021-09-15 by dhalbert