adafruit/circuitpython #5306: Bitbang SPI looses least significant bit

gingle, 2021-09-06:
### CircuitPython version

```python
Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit QT Py M0 with samd21e18 also
Adafruit CircuitPython 7.0.0-beta.0-161-g65753a1c2 on 2021-09-01; with samd51j19
```

### Code/REPL

```python
# 09/03/2021
# looses both least significant bits
# tested on QT Py and Metro_M4_Express

import board
import digitalio
import adafruit_bitbangio as bitbangio

nEN = digitalio.DigitalInOut(board.A1)    
nEN.direction = digitalio.Direction.OUTPUT

spi = bitbangio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)

def TestSPI():  
    select = nEN
    select.value = True
    while not spi.try_lock():
        pass
    spi.configure(baudrate=100000, polarity=0, phase=1, bits=8 )
    bytes_read = bytearray(4)
    select.value = False
    bytes_read = bytearray([0x33])  # changes 0x33 to 0x32 
    spi.write(bytes_read)
    bytes_read = bytearray([0x0b])  # changes 0x0b to 0x0a
    spi.write(bytes_read)
    select.value = True
    spi.unlock()
```

### Behavior

[image]

### Additional information

This works correctly:
from adafruit_bus_device.spi_device import SPIDevice

dhalbert, 2021-09-07:
Thanks for the report. A couple of questions:
1. Does your device require `phase=1`? Is it misbehaving?
2. Did you set the Saleae Logic SPI analyzer to CPHA=1? Screenshot below is from Logic 2.3.36:
[image]

The operative code is here, and looks OK to me on first glance:
https://github.com/adafruit/circuitpython/blob/d29a12cd81c12b76ce94317074c6396755ccde6c/shared-module/bitbangio/SPI.c#L152=L167

gingle, 2021-09-07:
> External Sender - Use caution opening files, clicking links, or responding to requests.

Hi Dan,

I first noticed this as I was driving a DDS chip. Using busio the DDS did not set to quite the right frequency. The LSBit is generally not a big impact but for the DDS it was the LSbit in the msb of the frequency set and set the frequency about 10% low over what it should have been. I was using the equivalent of the metro_m4_express. I then did a Busio version which worked OK with the same polarity/phase settings. I then tried the bitbang version on the QT PY which is the waveform that I included with the problem report. I tried both a 16 bit send and 2 eight bit sends. The two 8 bit sends dropped both LSbits. I am including the bitbang/SPI comparison waveforms.

[Inline image]

Regards,
Glen

[spi-phase-1]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F2847802%2F132354537-96a3d14c-0bc6-4cb2-991d-ca9a254cb000.png&data=04%7C01%7Cglen_ingle%40agilent.com%7C340b8704e0594109e86808d972053707%7Ca9c0bc098b46420693512ba12fb4a5c0%7C0%7C0%7C637666188992714807%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=6Htgbem34SLI3GbRng4Mhr%2FqRbQw69cUoD72K4Joro8%3D&reserved=0>

https://github.com/adafruit/circuitpython/blob/d29a12cd81c12b76ce94317074c6396755ccde6c/shared-module/bitbangio/SPI.c#L152=L167<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fadafruit%2Fcircuitpython%2Fblob%2Fd29a12cd81c12b76ce94317074c6396755ccde6c%2Fshared-module%2Fbitbangio%2FSPI.c%23L152%3DL167&data=04%7C01%7Cglen_ingle%40agilent.com%7C340b8704e0594109e86808d972053707%7Ca9c0bc098b46420693512ba12fb4a5c0%7C0%7C0%7C637666188992714807%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=fFBNUoNU6IavyakxpYBdBJ8PF%2FRJn6eqSTBmgrUNqOk%3D&reserved=0>

-

dhalbert, 2021-09-07:
Thanks, your inline image did not come through. Could you post directly in GitHub?
Which DDS chip is it?

[I edited your reply to remove the repetition of my post.]

gingle, 2021-09-07:
Hi Dan,

I am not well versed in Github unfortunately. One more thing to master. I posted the waveforms and programs on the Adafruit Forum under 'SPI Busio vs Bitbangio' on sept 3. It is under gingle in Adafruit Circuitpython and Micropython. The DDS was an AD9832.

Could you send any replies to ***@***.******@***.***>? I am doing most of my work at home and have my files on my home computer.

Thanks,
Glen

dhalbert, 2021-09-08:
I just realized you are using the https://github.com/adafruit/Adafruit_CircuitPython_BitbangIO Python-based library, not the built-in `bitbangio`module. Do you see the same problem when using native `import bitbangio`?

dhalbert, 2021-09-08:
> Could you send any replies to ***@***.******@***.***>? I am doing most of my work at home and have my files on my home computer.

We use GitHub so everyone can see the replies. When you get an email from GitHub, don't reply via mail, but instead click the "View it on GitHub" link in the email:

[image]

That will bring you directly here so you can reply.