adafruit/circuitpython #5307: Raw REPL greeting missing in CP 7.0 rc1

aivarannamaa, 2021-09-06:
### CircuitPython version

```python
7.0rc1 for RPi Pico
```

### Code/REPL

```python
# not related to code
```

### Behavior

In MicroPython up to latest release and in CircuitPython up to (at least) 6.3.0, entering the raw REPL gives fixed "greeting": `raw REPL; CTRL-B to exit>'`. In CP 7.0 rc1 for Raspbery Pi Pico, this greeting is lost and Thonny IDE can't read the confirmation about entering raw REPL anymore.

I haven't tested, but I suspect all tools depending on pyboard.py (eg. rshell and mpremote) will fail as well, because pyboard.py also checks for this marker: https://github.com/micropython/micropython/blob/bbbdef4cc1d2cb3a3f6c2d067b6988534773b2a6/tools/pyboard.py#L338

jepler, 2021-09-06:
It's possible something's buggy at 4a9ad8a9424bfdc16d58d69ce19bf4425296ff6d and this got lost?
```
-    mp_hal_stdout_tx_str("raw REPL; CTRL-B to exit\r\n");
+    serial_write_compressed(translate("raw REPL; CTRL-B to exit\n"));
```
in addition, this would only work in English, not in other languages once they've localized this message.

aivarannamaa, 2021-09-06:
> in addition, this would only work in English, not in other languages once they've localized this message.

I haven't checked it in 6.3.0, but in an older CP release this was not localized (and I hope it won't be, because raw repl is not meant for the users)

jepler, 2021-09-06:
Right, the ability to localize this string was added between 6.3.0 and now. Perhaps the impacts of doing so were not fully analyzed and it'll need to be reverted.  But that message looks a LOT like it's intended for a human to read and understand, which is one of the reasons I made it so that it could be localized.

jepler, 2021-09-06:
Can you test with that commit reverted and see if it fixes the problem you initially reported?

jwillikers, 2021-09-07:
I'm seeing this problem, too. I know this was not an issue during the alpha releases for 7.0.0.

aivarannamaa, 2021-09-07:
@jepler, reverting 4a9ad8a9424bfdc16d58d69ce19bf4425296ff6d makes CP usable in Thonny again.

I hope you agree to not localize `"raw REPL; CTRL-B to exit\n"`. The other strings in this commit are not relevant for Thonny and I agree that localizing Paste mode greeting makes sense.

jepler, 2021-09-07:
I see why localizing it is a problem for these other pieces of software right now, but a user can accidentally press the key combination to enter this mode.

We'll revert this change or a portion of it soon to avoid breaking other software, but can you help us plan how to make our possible to localize it in the future such as in circuit python 8?

aivarannamaa, 2021-09-07:
> can you help us plan how to make our possible to localize it in the future such as in circuit python 8?

The best I can think of is prefixing the standard marker with translated explanation, so that Ctrl+A would produce something like this:

```
NB! You have entered the mode which is meant for automatic tools, not for human users. Press Ctrl+B to get back to friendly shell
raw REPL; CTRL-B to exit
>
```

Current Thonny version would then output the first line as if it was regular output, but it would be able to detect raw REPL. In the future versions I could take care of the preceding line as well.

tannewt, 2021-09-07:
Are there ASCII or UTF-8 control characters we can use instead? I don't like prioritizing machine interoperability over human interop. Serial messages are *NOT* a stable API for software to work against.

aivarannamaa, 2021-09-08:
I wouldn't mind testing for fixed control characters in CircuitPython (and for the original marker in MicroPython), but I like my suggestion more, as pyboard.py would be happy with this and IMO it wouldn't hurt human interoperability at all.

tannewt, 2021-09-08:
In #5315 I have CircuitPython echo back the sent control codes to confirm what mode it is in. So, here is what it does:

* `\x01` (Ctrl-A) for raw repl. Sent after friendly message.
* `\x02` (Ctrl-B) for regular repl. Sent before friendly message when exiting paste or raw modes.
* `\x05` (Ctrl-E) for paste mode. Sent after friendly message.

aivarannamaa, 2021-09-08:
I'd vote for leaving #5315 for 8.0 and reverting to old behavior in 7.0.

aivarannamaa, 2021-09-08:
> I'd vote for leaving #5315 for 8.0 and reverting to old behavior in 7.0.

This would also leave more time to discuss it with MicroPython people. It would be nice if tool interface worked the same on both platforms.

dhalbert, 2021-09-08:
I agree it is late in the 7.0.0 cycle to change this, though we don't have to wait for 8.0.0: as soon as MicroPython and we agree, we can add it for some 7.x version.

@tannewt If you would like to provide a translatable message, for now we could print the English message _and_ print the `translate()` message if `TRANSLATION` is not English.

aivarannamaa, 2021-09-08:
> for now we could print the English message and print the `translate()` message if `TRANSLATION` is not English.

Please note, that tools currently look for `b"raw REPL; CTRL-B to exit\r\n>"`, so English *after* the translation would be better for tools.

closed 2021-09-08 by tannewt

tannewt, 2021-09-08:
Ok, I merged in @microDev1's revert and filed #5317 to track the work. Tools shouldn't be relying on message content so I'd rather put our efforts into determining invisible characters that serve the same purpose.

PR #5308 (2021-09-08): Revert localization of `REPL` messages
Closes #5307.