adafruit/circuitpython #5311: Mini Remote Control: No signal sent is valid?

eat-sleep-code, 2021-09-07:
@tannewt 

I am tinkering with parts from an old AdaBox -- a Circuit Playground Express and Mini Remote Control [Product ID: 389].

I installed CircuitPython 6.3.0 on the Circuit Playground Express. I also created a lib folder on the CPE and dropped the adafruit_irremote.mpy from Bundle for Version 6.x in it.

I then placed this code into a code.py file on it:

```
import board
import pulseio
import adafruit_irremote

pulseIn = pulseio.PulseIn(board.IR_RX, maxlen=120, idle_state=True)
decoder = adafruit_irremote.GenericDecode()

print("start")

while True:
    pulses = decoder.read_pulses(pulseIn)
    try:
        received = decoder.decode_bits(pulses)
    except adafruit_irremote.IRNECRepeatException:
        print("Repeat Exception")
        continue
    except adafruit_irremote.IRDecodeException:
        print("Decode Exception")
        continue
    except Exception as ex:
        print("General Exception", ex)
        continue
   
    print("Code received", received)
```

When I press the UP button (shown here pressed several times), this is the output.

> General Exception IRMessage(pulses=(6366, 251, 181, 241, 175, 191, 511, 249, 177, 273, 179, 214, 181, 271, 176, 2323, 175, 245, 181), reason='Pulses do not differ')
General Exception IRMessage(pulses=(179, 242, 174, 247, 179, 269, 178, 299, 179), reason='Too short')
General Exception IRMessage(pulses=(172,), reason='Too short')
Code received None
General Exception IRMessage(pulses=(178,), reason='Too short')
General Exception IRMessage(pulses=(179, 240, 176, 276, 197, 253, 205, 274, 179, 272, 175, 474, 181, 270, 177, 245, 181, 297, 311, 281, 181, 269, 178, 242, 174, 280, 178, 245, 182), reason='Pulses do not differ')
General Exception IRMessage(pulses=(174,), reason='Too short')
General Exception IRMessage(pulses=(179, 270, 177, 299, 179, 243, 172, 222, 173, 1685, 178, 724, 181, 1114, 176, 359, 182, 831, 178, 272, 175, 840, 206, 327, 177), reason='Only mark & space handled')

Not once did I appear to receive a valid code. 

I am not sure what is incorrect with my code -- for the most part it is pretty close to the example provided on the IR Test with Remote page.

DavePutz, 2021-09-08:
I tested the last script on CP 7.0.0-rc.1 and it worked correctly - no errors.

eat-sleep-code, 2021-09-09:
On 7.0.0-rc.1 I am getting a weird blink seqence on the Circuit Playgrounds.   Yellow flash of all the Neopixels and then two red flashes of all the neopixels, a pause and repeat of the red flashes.

tannewt, 2021-09-09:
@eat-sleep-code Those are the new status blinks for 7. The yellow flashes are the wait for safe mode on start up. The double red flashes indicate that the code.py ended due to an exception. The exception backtrace will show on the serial output.

eat-sleep-code, 2021-09-09:
@tannewt  Oh cool.   It was freezing in Thonny so never got to see what the exception was.   I will take a look at that again  a bit later today.

eat-sleep-code, 2021-09-10:
@tannewt What does a solid white ring indicate?

dhalbert, 2021-09-15:
I am moving this to the 7.x.x milestone since the bug was already present in 6.3.0.

closed 2021-10-12 by eat-sleep-code

eat-sleep-code, 2021-12-22:
This is still not resolved as of 7.1.0-beta.3

dhalbert, 2022-02-18:
I tried this with 3.x, 4.x, 5.x, and 6.x, and the last `adafruit_irremote` in each of the corresponding bundles.

The 5.x bundle `adafruit_irremote` (version 4.0.2 of the library) works with CircuitPython 6.3.0. The 4.1.1 version of the library fails with 6.3.0.

... AND, @lstein just fixed this yesterday with https://github.com/adafruit/Adafruit_CircuitPython_IRRemote/pull/51, which I didn't see until just now. Moving this issue back to the library.