adafruit/circuitpython #5318: nRF: deep sleep increases power usage

mrdalgaard, 2021-09-08:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-rc.1 on 2021-09-02; Adafruit Feather nRF52840 Express with nRF52840
```

### Code/REPL

```python
import time, alarm
time.sleep(15)
time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 30)
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```

### Behavior

After startup my cheap multimeter measures around 0.5mA for the first 15 seconds, and over twice that for the last 15 seconds (in deep sleep)
Light sleep and time.sleep has the same consumption, but deep sleep having a larger consumption is unexpected.

### Description

Easy workaround is to use

```
time.sleep(15)
import supervisor
supervisor.reload()
```
Instead of deep sleep, but deep sleep should be able to get the same consumption as light sleep.

hiredman, 2023-05-01:
I've recently run into this as well. Thought I would get my act together and use deepsleep, and it would give me a nice battery life boost in some ItsyBitsy nRF52840 Express based projects. Then I saw this report and ran some experiments. I get 1.5x to 2x the battery life just using time.sleep+supervisor.reload vs alarm.exit_and_deep_sleep_until_alarms.

I poked around a little looking at https://github.com/adafruit/circuitpython/tree/main/ports/nrf but have no idea where to start. A real bummer that deepsleep doesn't seem to be working correctly.

I'm now adding a 555 timer to those projects to cut power for some amount of time.

tannewt, 2023-05-02:
This is what implements deep sleep:
https://github.com/adafruit/circuitpython/blob/488dca565f700f40358eb3a6e50f1eb561f541c2/ports/nrf/common-hal/alarm/__init__.c#L263-L284

That file also includes light sleep.

hiredman, 2023-05-02:
wild guess:
https://github.com/adafruit/circuitpython/blob/488dca565f700f40358eb3a6e50f1eb561f541c2/ports/nrf/supervisor/port.c#L356-L397 appears to skip sleeping if there are any background callbacks, and I don't see anything in the loop here https://github.com/adafruit/circuitpython/blob/488dca565f700f40358eb3a6e50f1eb561f541c2/ports/nrf/common-hal/alarm/__init__.c#L182 that immediately looks like it is processing background callbacks. 
So maybe there is a background callback (dunno anything about cp internals and what those are or how they are used or if I can check for the presence of them from python), so it just spins in an a C loop until the timeout has elapsed instead of entering the deep sleep.

tannewt, 2023-05-03:
`RUN_BACKGROUND_TASKS;` should process the background callbacks.

hiredman, 2023-05-03:
yeah, missed that scanning for stuff followed by parens.

I wonder if it is softdevice related (which I guess is the radio?), if the softdevice is enabled it looks like it doesn't go into deepsleep, but instead waits for events from the sd

furbrain, 2023-05-03:
I've had a play and tried turning off the softdevice - `_bleio.adapter.enabled = False`. This doesn't have any effect

furbrain, 2023-05-03:
I can confirm that I get about 0.9 mA when just doing a simple timed sleep -

I'm guessing the reason why true deep sleep isn't used is due to needing a possible timed wake up. And SYSTEM OFF doesn't have any timers. Is there any reason why we couldn't use SYSTEM OFF if there were no timers?

hiredman, 2023-05-03:
Oh fun, when I do bleio.adapter.enabled = False in the repl running on the device, nothing happens. if I ctrl-d to exit the repl the device does a hard reset (serial port device disappears, lights blink, serial port device reappears). If I do the enabled=False in code.py without the repl, instant hard reset, and sometimes the device ends up in safe mode.

furbrain, 2023-05-03:
@hiredman  I don't suppose you are connecting to the REPL via bluetooth?

I've tried using `uctypes` to "manually" set up a pin alarm and deep sleep - and that seems to work really nicely, current consumption drops down to about 20uA as per my (fairly cheap) multimeter.
```py
  pin_no = get_pin_no(hardware.BUTTON_A)
  if pin_no >= 32:
      gpio_base_addr = 0x50000300 
      pin_no %= 32
  else:
      gpio_base_addr = 0x50000000
  pin_cnf_addr = gpio_base_addr + 0x700 + pin_no * 4
  set_uint32_at(pin_cnf_addr,0x0003000C) # sense low, input buffer, pullup
  set_uint32_at(0x40000500, 1) # SYSTEM OFF
```

furbrain, 2023-05-03:
Obviously this doesn't help if you are wanting it to wake after a specific time

hiredman, 2023-05-03:
nope, repl over serial over usb.

tannewt, 2023-05-04:
> Is there any reason why we couldn't use SYSTEM OFF if there were no timers?

How would you wake it from system off?

furbrain, 2023-05-04:
@tannewt responding to gpio changes. In the application I am working on, the user presses a button to turn the device on. Also connecting Vbus wakes the system.

Sola85, 2024-08-15:
This might be completely unrelated, but I was also debugging why my nrf draws around 0.5-2 mA in deep sleep. I had detachted the USB data connection, so I was measuring true deep sleep and not fake deep sleep. My mistake was that while I had disconnected USB data, USB power was still connected. The mcu appears to draw ~1mA from VDD in sleep mode if VBUS is at 5V. 

If I detach VBUS, I get ~ 2µA deep sleep with only time alarms and around 20 µA with a pin alarm active.