adafruit/circuitpython #5325: Vectorio doesn't consistently validate location

lesamouraipourpre, 2021-09-09:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-beta.0 on 2021-08-24; Adafruit PyPortal with samd51j20
```

### Code/REPL

```python
import displayio
import vectorio

line_palette = displayio.Palette(1)
line_palette[0] = 0xFFFFFF

line_one = vectorio.Rectangle(width=64, height=1, x=0, y=0, pixel_shader=line_palette)

# Test Three - Doesn't error when location is outside of signed int 16 values
line_one = vectorio.Rectangle(width=64, height=1, x=50000, y=50000, pixel_shader=line_palette)
print("Test One: Location set to (50000, 50000) but stored as", line_one.location, "It should have errored")

for i in range(0,100000,10000):
    # Test Two - Doesn't error when location is outside of signed int 16 values
    line_one.x = i
    line_one.y = i
    print("Test Two", i, line_one.location)

for i in range(0,100000,10000):
    print("Test Three", i)
    # Test Three - Correctly errors when location is outside of signed int 16 values
    line_one.location = (i, i)
    print("  T3", line_one.location)
```

### Behavior

```python
Test One: Location set to (50000, 50000) but stored as (-15536, -15536) It should have errored
Test Two 0 (0, 0)
Test Two 10000 (10000, 10000)
Test Two 20000 (20000, 20000)
Test Two 30000 (30000, 30000)
Test Two 40000 (-25536, -25536)
Test Two 50000 (-15536, -15536)
Test Two 60000 (-5536, -5536)
Test Two 70000 (4464, 4464)
Test Two 80000 (14464, 14464)
Test Two 90000 (24464, 24464)
Test Three 0
  T3 (0, 0)
Test Three 10000
  T3 (10000, 10000)
Test Three 20000
  T3 (20000, 20000)
Test Three 30000
  T3 (30000, 30000)
Test Three 40000
Traceback (most recent call last):
  File "<stdin>", line 23, in <module>
ValueError: unsupported point type
```

### Description

When you set the location of a `vectorio` shape via the constructor, the `x` or `y` properties, no bounds checking is performed. If it is set via the `location` property, bounds checking is performed and raises a `ValueError`

lesamouraipourpre, 2021-09-09:
Looking at this further the constructor expects the x,y location to be in `uint16` range but the validation is done against `sint16` range.

https://github.com/adafruit/circuitpython/blob/0ee0ed70350f029c0ebf05ceaeb89d5611ffa297/shared-module/vectorio/VectorShape.c#L165-L167

tannewt, 2021-09-10:
Let's not block 7.0.0 on this. It can be fixed afterwards.

closed 2021-09-11 by microDev1

WarriorOfWire, 2021-09-11:
@tannewt agree this shouldn't slow down the 7.0.0 release train. Good candidate for 7.0.1 inclusion.

PR #5327 (2021-09-11): Consistently validate location on VectorShape
Closes #5325

The x, y or location properties now use the same bounds validation whether set via the constructor or setters.
Tested with the code in #5325 which now raises Errors consistently when out of range.

@WarriorOfWire can you check if these changes are consistent with the rewrite and pass any test code you have?