adafruit/circuitpython #5331: Implement CPU sleep (WFI) when idle on RP2040

urish, 2021-09-11:
Looking at the rp2 port code, I can see that there's already an [implementation for `port_idle_until_interrupt()`](https://github.com/adafruit/circuitpython/blob/cf5c32be3b3708035271ee264f978207ce928dd1/ports/raspberrypi/supervisor/port.c#L238), but for some reason the `wfi` instruction is commented out.

When running CircuitPython in the simulator, this makes a huge difference, as you can see:
* [Blink with stock CircuitPython 7.0.0-rc.1](https://wokwi.com/arduino/projects/309474946192507458)
* [Blink with patched CircuitPython, uncommenting the `__wfi()` call in port_idle_until_interrupt](https://wokwi.com/arduino/projects/309477633211499073)

Is there any blocker for using `wfi` on the rp2 port? Is there anything I can do to help with this?

tannewt, 2021-09-13:
I had copied it in there from SAMD21 when doing the initial port work. It didn't work for some reason so I commented it out and moved on.

If it does work now, then great! A PR would be welcome. Please test on real hardware too.

I don't have any blocker for it. I just never circled back to retest it.

urish, 2021-09-13:
Thanks for explaining! I'll try to see how this works on the real hardware and send a PR. 

One thing I noticed - it doesn't work (in the simulator) unless I also comment out the call to `common_hal_mcu_disable_interrupts()` before calling WFI. 

What's the reason we disable interrupts there?

tannewt, 2021-09-13:
Without it there is a chance an interrupt occurs, is serviced and queues a background task before the WFI. Doing the WFI then delays the execution of the background task and messes things up.

WFI should wake from an interrupt even when interrupts aren't enabled. This allows shutdown and wake code to happen around an WFI before needing to service an interrupt.

urish, 2021-09-13:
Got it, thanks!

urish, 2021-09-13:
Ok, so I ran a build with the two lines uncommented, and it seems to pass a quick sanity test:

1. The system loads, I can see the CP drive
2. I have a serial console (over USB)
3. I can copy my code to the CP drive, it runs
4. The code blinks the LED successfully and prints to the Serial output
5. I can get into the REPL and type commands successfully

I used the following test program for step 4:

```python
import board
import digitalio
import time

led = digitalio.DigitalInOut(board.GP25)
led.direction = digitalio.Direction.OUTPUT

counter = 0
while True:
  print("Blink! {}".format(counter))
  led.value = True
  time.sleep(0.5)
  led.value = False
  time.sleep(0.5)
  counter += 1
```

What else should I test before making a PR?

dhalbert, 2021-09-13:
Were you testing on a real RP2040 board?

urish, 2021-09-13:
Yes, like @tannewt asked. In fact, it fails right now in the simulator because of the WFI while interrupts disabled issue. But I'm working on a fix for it now

microDev1, 2021-09-16:
Merging PR #5341 caused issue #5354 so it was reverted in #5357.

PR #7201 (2022-11-15): raspberrypi: Make port_idle_until_interrupt work
This needs thorough testing before it's merged, as we tried and reverted this once before (#5341 and #5356).

I think that besides checking for tinyusb having "something to do", the fact that `port_interrupt_after_ticks` and `port_disable_tick` weren't implemented that was causing a secondary problem.

I've tested this on a pico w over reboot-cycles and ctrl-c-cycles, with and without drive automounting, with and without serial repl open, and on a power-only connection.

I didn't notice the problem reported in #5356 after merely implementing por [... cut]