adafruit/circuitpython #5343: alarm.wake_alarm not set upon wake from deep sleep

aocole, 2021-09-14:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-rc.1-62-gbe7d963a9-dirty on 2021-09-12; Adafruit MagTag with ESP32S2
```

### Code/REPL

```python
import alarm
import board
import digitalio
import neopixel
import time
import microcontroller
from adafruit_magtag.magtag import MagTag

print("Hello World!")

magtag = MagTag()
magtag.add_text(
    text_scale=2,
    text_wrap=25,
    text_maxlen=300,
    text_position=(10, 10),
    text_anchor_point=(0, 0),
)
pause = alarm.time.TimeAlarm(
    monotonic_time=time.monotonic() + 60
)
magtag.set_text(
    "battery: {}V     wake_alarm: {}".format(
        magtag.peripherals.battery, repr(alarm.wake_alarm)
    )
)
magtag.peripherals.deinit()
print("Sleeping in 5")
time.sleep(5)
print("sleeping..")

magtag.refresh()
button = alarm.pin.PinAlarm(pin=board.D15, value=False, pull=True)
alarm.exit_and_deep_sleep_until_alarms(pause, button)
```

### Behavior

When plugged in to USB (and so the board only pretends to deep sleep), alarm.wake_alarm is set as expected to an instance of either TimeAlarm or PinAlarm. However, when running on battery, alarm.wake_alarm is always None when the above code runs upon wake, regardless of whether it was the pin or timer alarm that caused the wake. 

### Additional information

Thanks to @KeithTheEE for helping to validate this with me on discord.

KeithTheEE, 2021-09-14:
Using most of the above code I tested with an adjustment that uses the `isinstance` comparison as outlined in [this learn guide](https://learn.adafruit.com/deep-sleep-with-circuitpython/alarms-and-sleep#what-woke-me-up-3079890-16) which states that the deep sleep should at least should set the `alarm.wake_alarm` to the correct alarm type, and an `isinstance` comparison should give you an idea of which alarm woke it thanks to that. 

```py

magtag.set_text(
    "battery: {}V     Pin: {}             Timer: {}".format(
        magtag.peripherals.battery, str(isinstance(alarm.wake_alarm, alarm.pin.PinAlarm)),
        str(isinstance(alarm.wake_alarm, alarm.time.TimeAlarm))
    )
) 
```

Running the code with this adjustment, wake_alarm is neither type. This isn't a surprise as @aocole's code shows that it's NoneType, but just for sanity's sake I figured I'd start there.

anecdata, 2021-11-23:
I don't know if this is a different issue, or if the current issue is more complex.

Testing with:
`Adafruit CircuitPython 7.1.0-beta.0-85-g387a8a46b on 2021-11-22; Adafruit MagTag with ESP32S2`

in code.py:
```
print(alarm.wake_alarm, microcontroller.cpu.reset_reason, supervisor.runtime.run_reason)
```

If I've been in simulated since power-on, I get:
`None microcontroller.ResetReason.POWER_ON supervisor.RunReason.STARTUP`
 on power-on, then on subsequent sleep cycles:  
`<TimeAlarm> microcontroller.ResetReason.POWER_ON supervisor.RunReason.STARTUP`
 
But if I come back to connected from unconnected, I get: 
`None microcontroller.ResetReason.DEEP_SLEEP_ALARM supervisor.RunReason.STARTUP the first time `
on power-on, then on subsequent sleep cycles:  
`<TimeAlarm> microcontroller.ResetReason.DEEP_SLEEP_ALARM supervisor.RunReason.STARTUP `

tbjers, 2021-12-16:
Yup, this is definitely a thing. I found that if I do something along these lines:

```python
def deep_sleep():
    print("Sleeping for %d minutes" % REFRESH_TIME)
    magtag.peripherals.neopixel_disable = True
    magtag.peripherals.speaker_disable = True
    alarm.exit_and_deep_sleep_until_alarms(pin_alarm, time_alarm)

# Set up alarms for the different buttons and timer
pin_alarm = alarm.pin.PinAlarm(board.D14, value=False)
time_alarm = alarm.time.TimeAlarm(monotonic_time=int(time.monotonic()) + (REFRESH_TIME * 60))

# Create a new MagTag object
magtag = MagTag()
r = rtc.RTC()

# noinspection PyProtectedMember
secrets = magtag.network._secrets
```

Then at least button D14 will wake the MagTag up, just won't wake it up with an actual alarm, as stated.

tbjers, 2021-12-16:
To further shed some light on this, I found that I can access the type of `alarm.wake_alarm` from `boot.py`.

This code:

```python
# SPDX-FileCopyrightText: 2021 Torgny Bjers, written for THE LULZ
#
# SPDX-License-Identifier: Unlicense

import alarm
import storage
import time

if alarm.wake_alarm:
    try:
        storage.remount("/", readonly=False)
        with open("/log.txt", "a") as fp:
            test = dir(alarm.wake_alarm)
            for s in test:
                fp.write(f'{s}\n')
            fp.flush()
            time.sleep(1)
            storage.remount("/", readonly=True)
            storage.enable_usb_drive()
    except OSError as e:
        print("Something died.")
```
Yields the following in the text file `log.txt`:
```python
__class__
value
pin
```

dhalbert, 2021-12-22:
Fixed by #5762.

closed 2021-12-22 by dhalbert

tbjers, 2021-12-22:
Thank you so much for fixing this!

patja, 2021-12-23:
Yes thank you so much for this fix.  It cleared the path to completing some Christmas gift projects that were confounding me!

pookzilla, 2022-07-17:
This still appears to be an issue in 8.0.0alpha1 on an Adafruit Feather ESP32-S3 TFT.  alarm.wake_alarm consistently has the correct value when powered by USB but is consistently None when on battery power.

PR #5762 (2021-12-22): Fix recording of wake_alarm
- Fixes #5343.

`alarm.wake_alarm` was not being set properly after a real or fake deep sleep. I moved the responsibility for keeping track of whether the board had just hard-restarted to `main.c`, instead of having it in `common-hal`. The most recent wake alarm is now recorded in `alarm.wake_alarm` either after a hard-reset, or after a simulated (fake) deep sleep.

Light sleep was not setting `alarm.wake_alarm`. I did this in a port-independent way.

Also did a minor doc fix in `alarm`.

It looks like `circuitpython.pot` got out of sync with the code base at some p [... cut]