adafruit/circuitpython #5344: stm32f405 has sdioio module as sdio

jerryneedell, 2021-09-14:
Was the name of sdioio changed to sdio recently? the docs still refer to it as sdioio but in the rc.1 build for the STM32F405, it shows up (and works) as sdio instead of sdioio.

Is this correct?

```
Adafruit CircuitPython 7.0.0-rc.1-72-g9b2d8ade2 on 2021-09-14; Adafruit Feather STM32F405 Express with STM32F405RG
>>> 
>>> help('modules')
__main__          board             math              sharpdisplay
_bleio            builtins          microcontroller   storage
adafruit_bus_device                 busio             micropython       struct
adafruit_pixelbuf canio             msgpack           supervisor
aesio             collections       neopixel_write    synthio
alarm             digitalio         onewireio         sys
analogio          displayio         os                terminalio
array             errno             pulseio           time
atexit            fontio            pwmio             touchio
audiocore         framebufferio     rainbowio         traceback
audiomp3          gc                random            ulab
audiopwmio        getpass           re                usb_cdc
binascii          io                rgbmatrix         usb_hid
bitbangio         json              sdcardio          usb_midi
bitmaptools       keypad            sdio              vectorio
Plus any modules on the filesystem
>>>

microDev1, 2021-09-14:
Good catch @jerryneedell. This change was introduced in https://github.com/adafruit/circuitpython/commit/9002f351b46f0dce620e2dcaf3c9550694c777dc, it is probably a mistake.

jerryneedell, 2021-09-14:
> Good catch @jerryneedell. This change was introduced in [9002f35](https://github.com/adafruit/circuitpython/commit/9002f351b46f0dce620e2dcaf3c9550694c777dc), it is probably a mistake.

Ah -- good -- I've been trying to find another port that has it enabled to compare, but have not found one yet. IT is in the "common-hal" for atmel-samd but it appears to be disabled in the build.

jerryneedell, 2021-09-14:
Is it just this line https://github.com/adafruit/circuitpython/blob/9002f351b46f0dce620e2dcaf3c9550694c777dc/shared-bindings/sdioio/__init__.c#L49
missing the "io" in the QSTR_sdio

microDev1, 2021-09-14:
Yup, that is the only line that needs to be changed.

```diff
- MP_REGISTER_MODULE(MP_QSTR_sdio, sdioio_module, CIRCUITPY_SDIOIO);
+ MP_REGISTER_MODULE(MP_QSTR_sdioio, sdioio_module, CIRCUITPY_SDIOIO);
```

jerryneedell, 2021-09-14:
I can do a PR later today, but it may be a lot quicker for someone else to do it.

dhalbert, 2021-09-14:
I just did a quick PR.

jerryneedell, 2021-09-14:
Thanks -- it works!

PR #5345 (2021-09-14): Restore sdioio name (was sdio)
- Fixes #5344.

Just a typo fix.
closed 2021-09-14 by jerryneedell