adafruit/circuitpython #5346: SSD1309 Display Library

Capekk, 2021-09-14:


tannewt, 2021-09-15:
It should be doable. I'd start with this driver: https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_SSD1306

What display(s) are you using?

Capekk, 2021-09-15:
It's 2.42 inch 128x64 OLED LCD Display Module SSD1309 12864 7 Pin SPI/IIC I2C

deshipu, 2021-09-15:
Try and see if the ssd1306 driver works, it should be similar enough.

Capekk, 2021-09-16:
Ye I'm gonna try it, but I haven't received it yet. I was just asking if it's even possible. Thanks for the help guys.

dhalbert, 2024-03-08:
@Capekk This is an ancient issue, but did you get the SSD1309 to work with the SSD1306 driver?

Capekk, 2024-03-08:
> issue

Hello @dhalbert, 
I did! In the end I used adafruit_ssd1306 and adafruit_framebuf libraries for my project. The setup was set like this: 
oled = adafruit_ssd1306.SSD1306_I2C(128, 64, i2c)
It worked without issues. I don't think that the latter library is even necessary, since I've only used it for the graphics... Hope it helps.

dhalbert, 2024-03-08:
Great! Glad to hear that it worked!

closed 2024-03-08 by dhalbert