adafruit/circuitpython #5351: matrixportal_m4 build errors with arm-none-eabi-gcc 11.2.0

rdoursenaud, 2021-09-15:
### CircuitPython version

```python
7.0.0-rc.2
```

### Code/REPL

```python
NA
```

### Behavior

```[raph@STANVII atmel-samd]$ make -j17 BOARD=matrixportal_m4
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
QSTR updated
../../devices/ble_hci/common-hal/_bleio/Adapter.c: In function 'bleio_adapter_gc_collect':
../../devices/ble_hci/common-hal/_bleio/Adapter.c:933:75: error: expression does not compute the number of elements in this array; element type is 'bleio_connection_internal_t', not 'size_t' {aka 'unsigned int'} [-Werror=sizeof-array-div]
  933 |     gc_collect_root((void **)bleio_connections, sizeof(bleio_connections) / sizeof(size_t));
      |                                                                           ^
../../devices/ble_hci/common-hal/_bleio/Adapter.c:933:75: note: add parentheses around the second 'sizeof' to silence this warning
../../devices/ble_hci/common-hal/_bleio/Adapter.c:74:29: note: array 'bleio_connections' declared here
   74 | bleio_connection_internal_t bleio_connections[BLEIO_TOTAL_CONNECTION_COUNT];
      |                             ^~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:74: build-matrixportal_m4/common-hal/_bleio/Adapter.o] Error 1
make: *** Waiting for unfinished jobs....
```

### Description

Using the latest upstream release the build fails.
``` 
arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 11.2.0
```

### Additional information

Building with the officially recommended v10.2.1 works fine.

tannewt, 2021-09-16:
I think I did `add parentheses around the second 'sizeof' to silence this warning` locally. Want to make a PR for it?

closed 2022-03-09 by dhalbert

dhalbert, 2022-03-18:
The underlying issue is being discussed in https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578, as mentioned in the code. It looks like the gcc folks are fixing the warning regression, which got a lot of pushback from people trying to access memory at fixed addresses, e.g. `(void *) 0x12345678`.

PR #6128 (2022-03-09): Fix SAMD51 builds on GCC11.2
Fixes #5351