adafruit/circuitpython #5354: Macropad and other rp2 boot up issues with RC2

Neradoc, 2021-09-16:
Updating a macropad to RC2 from the site made the board not mount (and make the mouse stutter).
When I plug the board or reset, the safe mode blink happens, then nothing. Not even the screen shows anything.
Putting RC1 back, and power cycling restored it to normal.

Trying recent builds, I narrowed it down to #5341. Reverting it in RC2 fixed the issue.

On a Raspberry pico, RC2 does not show or run the code when plugged in, but comes back and seems to run ok after pressing reset. The fix makes it run as expected when plugged in. Same thing with a Pimoroni Keybow 2040 or QT PY 2040.

dhalbert, 2021-09-16:
I reproduced this on a Feather RP2040. On power-up, nothing is seen. I wrote a blink program in `code.py` and it did not start. Pressing reset usually causes the board to come up, but I had to switch USB ports after a second power-cycle, so I think the USB behavior is upsetting Linux.

dhalbert, 2021-09-16:
I tried adding an explicit check for `!tud_task_even_ready()` before the `__wfi()`, which is what the code did before #5182, and that did not help.

@tannewt Since this is a show-stopper, we could just revert #5341, and I'll make an rc.3 quickly. Let's talk about what to do next.

jpconstantineau, 2021-09-16:
I can confirm this issue is present on another RP2040 board I am working on (not yet in the list but loaded a Pico uf2 as a first test - received the PCBs before dinner...).

PR #5357 (2021-09-16): Revert #5341. Does not work on power-cycle.
Reverts #5341 to fix #5354 for now. We'll do a more detailed diagnosis after 7.0.0 final.

Discussed in terms of scheduling with @tannewt.
closed 2021-09-16 by dhalbert