adafruit/circuitpython #5355: Application using USB HID on ESP32S2 hangs μC when executed out of code.py but executes properly from REPL

evildave666, 2021-09-16:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-rc.2 on 2021-09-15; S2Mini with ESP32S2-S2FN4R2
```

### Code/REPL

```python
import board
import digitalio
from adafruit_debouncer import Debouncer
import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode

pin = digitalio.DigitalInOut(board.D2)
pin.direction = digitalio.Direction.INPUT
pin.pull = digitalio.Pull.UP
switch = Debouncer(pin)

led = digitalio.DigitalInOut(board.D4)
led.direction = digitalio.Direction.OUTPUT

keyboard = Keyboard(usb_hid.devices)

while True:
    switch.update()
    led.value = switch.value
    if switch.fell:
        keyboard.send(Keycode.Z)
```

### Behavior

When the code above (in /keeb.py) is run out of REPL manually via ">>> import keeb" it runs fine (extinguishes the LED and sends "z" to host when key is pressed)

When a code.py consisting solely of:
```python
import keeb
```
is created, no CDC device is attached to the host for REPL, CIRCUITPY is not mounted, and the program does not function.
The device can only be brought back to normalcy via safe mode and removing the code.py.

### Description

- Behaviour not occur when an RP2040 device is substituted for the ESP32S2-FN4R2 device I'm using (I do not have any other ESP32 devices to test with)
- Occurs regardless of presence of a boot.py
- This started occurring prior to rc.2, I believe on an interim build between an alpha and rc.1
- When all HID code is commented out from keeb.py it starts up properly from code.py (extinguishing the LED when the key is pressed but obviously not sending the keycode back)

### Additional information

Attached hardware: normal open momentary keyswitch (cherry mx clone) between D2 (IO33) and ground, red led w/ 470ohm resistor between D4 (IO17) and ground.

dhalbert, 2021-09-16:
Hypothesis: we do a gc after each `import`, so I wonder if something HID-related is being gc'd that should not be.

dhalbert, 2021-09-16:
I tried reproducing this on a Metro ESP32-S2, using `board.IO5` for the button and `board.LED` for the LED. Otherwise everything is the same. Libraries are all the latest versions. There is no `boot.py`, and `keeb.py` and `code.py` are as described above.

I do not see this problem with rc.2, including when restarting from a power-cycle or a ctrl-C, ctrl-D in the REPL.

dhalbert, 2021-09-16:
Further testing: no problem with rc.2 .uf2 instead of  .bin. @Neradoc can also not replicate on TinyS2 or Feather S2.

durapensa, 2021-09-17:
Tested & no problem with rc.3 .uf2 (using `board.BUTTON` instead of `board.D2`)

evildave666, 2021-09-18:
OK, since @durapensa is having no problem with the same board I guess we can safely assume this board is probably naff. Sorry to waste time on this.

closed 2021-09-18 by evildave666