adafruit/circuitpython #5359: changed file checking is breaking builds in my fork

jepler, 2021-09-16:
### CircuitPython version

```python
7.0.0-rc.3
```

### Code/REPL

```python
#
```

### Behavior

github actions fails [in my fork](https://github.com/jepler/circuitpython/runs/3625108063?check_suite_focus=true#step:35:7):
[image]
```
Run python3 -u ci_changed_board_list.py
Error: Argument list too long
```

The value of CHANGED_FILES is very lengthy. 
[image]

I do not know why it's so long (a bug in dorny/paths-filter@v2?), but this could also occur in a legitimate very large PR (such as one revising the code formatting of a large number of files, merging micropython, etc), so it doesn't work to stick it all in an environment variable.

### Additional information

on my _local_ system, an environment variable setting fails if the whole `NAME=value` part is over 131071 bytes. The `NAME=value` show in my pull request is 179115 bytes.

tannewt, 2021-09-17:
I couldn't find another way to pass the info from one step to another.

I think this is more a bug in the base branch detection. It determined that this was a new branch so it diffed against your main branch which may not be up to date. There is likely some setting we could tweak to change this.
[image]

jepler, 2021-09-22:
A possible alternate syntax that does not require the changed files list to be an environment variable.
```yaml
        steps:

        - uses: actions/checkout@v2
        - uses: dorny/paths-filter@v2
          id: filter
          with:
            # Enable listing of files matching each filter.
            # Paths to files will be available in `${FILTER_NAME}_files` output variable.
            # Paths will be formatted as JSON array
            list-files: json

            # Compare against this branch. (Ignored for PRs.)
            base: ${{ github.ref }}

            # In this example all changed files are passed to the following action to do
            # some custom processing.
            filters: |
              changed:
                - '**'
        - name: use the paths
          run: |
            cat <<<'${{ steps.filter.outputs.changed_files }}'
```

With the above in a personal repo, I was able to successfully run on a change of 13107 files, with the total size of the json data over half a megabyte.  However, the script needs to be changed to use json data on stdin instead of an environment variable.

PR #7708 (2023-03-14): Use intersection of changes per commit and merge ref
This will filter out changes introduced by merge commits (merging pr base branch into head branch) inside the commit range.
Also, resolves #5359 by computing changes inside the script.
closed 2023-03-14 by tannewt