adafruit/circuitpython #5370: json module can not round

jddmartin, 2021-09-18:
### CircuitPython version

```python
Adafruit CircuitPython 6.3.0 on 2021-06-01; Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
"""minimal working example of bug"""

import json

x = 5e6  # 5e5 works

old_d = {"number":x}

print("old_d", old_d)

jstring = json.dumps(old_d)

print("json string: ", jstring)

new_d = json.loads(jstring)

print("new_d", new_d)
```

### Behavior

code.py output:
old_d {'number': 5e+06}
json string:  {"number": 5e+06}
Traceback (most recent call last):
  File "code.py", line 15, in <module>
ValueError: invalid syntax for number

### Description

Floating point numbers of the form 5e+06 are generated by "json.dumps" but not understood by "json.loads"

"json.loads" understand 5e6 though.

jepler, 2021-09-19:
If possible, please test with the latest 7.0.0 release candidate.

If possible, please test again with the latest version, 7.0.0-rc.3 or newer.  It appears it was fixed somewhere along the line.
```
Adafruit CircuitPython 7.0.0-rc.3-dirty on 2021-09-16; Adafruit Macropad RP2040 with rp2040
>>> import json
>>> json.loads('{"number": 5e+06}')
{'number': 5e+06}
```

jddmartin, 2021-09-19:
After updating to 7.0.0-rc.3.uf2 I no longer see the bug.

Thank you!

closed 2021-09-19 by jddmartin