adafruit/circuitpython #5377: Crash into HardFault_Handler when importing the built-in wifi module on Saola 1 w/Wrover with ESP32S2

myllyja, 2021-09-21:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; Saola 1 w/Wrover with ESP32S2
Board ID:espressif_saola_1_wrover
```

### Code/REPL

```python
import wifi
```

### Behavior

If the code.py tries to import the wifi module, the device will go to safe mode

```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
```

And if I try to do the same in REPL, the serial connection is cut off(PuTTY says "Error reading from serial device") without any further output after entering the command.
```
Adafruit CircuitPython 7.0.0 on 2021-09-20; Saola 1 w/Wrover with ESP32S2
>>> import wifi
```

### Additional information

I started from 6.3.0 where the wifi module could be loaded but it was unable to scan for any networks so I decided to try updating to CircuitPython 7.0.0. I've since narrowed the crash to start happening somewhere between 7.0.0-alpha.3 and 7.0.0-alpha.4.

On 7.0.0-alpha.3 and before the wifi module can be imported without a crash, but it's still unable to scan any networks. Starting from 7.0.0-alpha.4 the device crashes when importing the built-in module.

I've also tested with the latest binary "Adafruit CircuitPython 7.0.0-62-g9689022fb on 2021-09-21; Saola 1 w/Wrover with ESP32S2" and the same crash occurs.

dhalbert, 2021-09-21:
Are you using the .uf2 or the .bin version of CircuitPython?

myllyja, 2021-09-21:
I started with the .uf2 of 7.0.0 and then the .bin of 7.0.0, same issue. Further testing has been done with the .bin versions from https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/espressif_saola_1_wrover/en_US/ using the https://adafruit.github.io/Adafruit_WebSerial_ESPTool/ site to write the firmware bins.

anecdata, 2021-09-21:
I did a clean install of 7.0.0 on the MagTag and the FeatherS2 yesterday. Install sequence: command line `erase_flash`, command line install TinyUF2, install (drag) CircuitPython UF2, install libraries and `.py` files, ctrl-C & return in console to enter REPL, EJECT Volume, hard reset, then let it run.

Saw 1st hardfault on one of these devices this morning.

dhalbert, 2021-09-21:
Do you have a USB breakout connected to D19 and D20? Could you try erasing the flash and then loading the .bin, or did you try that already? If you switch between the .uf2 and .bin, you need to erase the flash in between.

myllyja, 2021-09-21:
Yes, I have a USB breakout connected to D19, D20, 5V and GND. The initial update to 7.0.0 was done by using the process described in https://circuitpython.org/board/espressif_saola_1_wrover/ to Install/Update the UF2 bootloader with the Adafruit WebSerial ESPTool. First the Erase, then writing the combined.bin and once the USB drive SAOLA1BOOT shows up, copied the adafruit-circuitpython-espressif_saola_1_wrover-en_US-7.0.0.uf2 on it and once it's back up the "Hello World" prints normally, but trying to load the wifi module in REPL or code.py crashes the device. Afterwards I've skipped the UF2 and just used the WebSerial to always do an Erase before uploading the bin file.

dhalbert, 2021-09-21:
This is working for me, not using the WebSerial tool. (I am not having a lot of luck with the WebSerial tool for some reason.)

```sh
$ esptool.py --chip esp32s2 -p /dev/ttyACM0 -b 460800 erase_flash
esptool.py v3.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:00:7d:cc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.6s
ERROR: ESP32-S2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this error, set --after option to 'no_reset'.

# [pressed BOOT, then RST, then release both]

$ esptool.py --chip esp32s2 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=no_reset write_flash 0x0000 adafruit-circuitpython-espressif_saola_1_wrover-en_US-7.0.0.bin 
esptool.py v3.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:00:7d:cc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00151fff...
Compressed 1383792 bytes to 889517...
Wrote 1383792 bytes (889517 compressed) at 0x00000000 in 12.7 seconds (effective 868.4 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

# [pressed RST]

$ tio /dev/ttyACM0
[tio 15:31:37] tio v1.32
[tio 15:31:37] Press ctrl-t q to quit
[tio 15:31:37] Connected
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0 on 2021-09-20; Saola 1 w/Wrover with ESP32S2
>>> import wifi
>>> 
```

dhalbert, 2021-09-21:
BTW, I am testing with an `ESP32-S2-Saoloa-1_V1.2`, as labelled on the bottom of the board. Is your board version different?

myllyja, 2021-09-21:
I still have the same crash when writing with esptool.py
```
# hold BOOT, click RST, release BOOT
PS C:\circuitpython7\firmware> esptool.py --chip esp32s2 --port COM12 -b 460800 erase_flash
esptool.py v3.0
Serial port COM12
Connecting...
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:0e:e1:3c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 17.3s
Hard resetting via RTS pin...
ERROR: ESP32-S2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this error, set --after option to 'no_reset'.

# hold BOOT, click RST, release BOOT

PS C:\circuitpython7\firmware> esptool.py --chip esp32s2 --port COM12 -b 460800 --before=default_reset --after=no_reset write_flash 0x0000 .\adafruit-circuitpython-espressif_saola_1_wrover-en_US-7.0.0.bin
esptool.py v3.0
Serial port COM12
Connecting...
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:0e:e1:3c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 1383792 bytes to 889517...
Wrote 1383792 bytes (889517 compressed) at 0x00000000 in 21.8 seconds (effective 506.9 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

# [pressed RST]

PS C:\circuitpython7\firmware> putty -serial COM13
```
[image]

Yes, same ESP32-S2-Saola-1_V1.2 marking on the bottom. I'm using two breadboards side by side, one without the power rails to get to all the pins on it.

Oh crap, while I was looking at the bottom of the board to see the version I tested without the rest of the project connected and the wifi started working.. I then found out that I had wired two input pins into GPIO4 and 5 from a 5V device(measured 4.0 V and 4.8 V on the pins 😨). When I moved that external component from 5V to a 3.3V source, everything started working again. So problem solved! Unless someone wants to dig deeper into why it actually crashes starting from a certain CircuitPython version.

dhalbert, 2021-09-21:
> I then found out that I had wired two input pins into GPIO4 and 5 from a 5V device(measured 4.0 V and 4.8 V on the pins ). When I moved that external component from 5V to a 3.3V source, everything [...]

I don't know why, but the pin voltage ratings are no more than 0.3V above Vdd, and operation is undefined above those. It could that there was a problem caused that just happened not to affect wifi, and then it did, due to code size changes, things moving around, etc.

I will close this for now, but of course let us know if there's something reproducible with rated operation. Thanks for your persistence in trying various things.

closed 2021-09-21 by dhalbert