adafruit/circuitpython #5380: How to enable keyboard to wake sleeping computer?

LostVector, 2021-09-22:
Hi,

I'm trying to figure out how to make an emulated USB keyboard (using a neokey 1x4) wake the computer it's attached to.  I've verified the QT PY RP2040 is getting power and behaves normally when the computer is on, but it seems to go haywire when the pc is asleep and i press a key (windows 10).

Is this possible to support somehow with the current libraries?

dhalbert, 2021-09-22:
Transferring to https://github.com/adafruit/circuitpython, as this is a core issue.

dhalbert, 2021-09-22:
@hathach What would we need to do to make USB wake work? I assume we are not doing USB at all when we detect that the host has gone away (disconnect or suspend, I guess, I'm not sure which). Thanks.

jpconstantineau, 2021-09-22:
Just a pointer to what perhaps does it...  It's arduino code example on Tinyusb: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/HID/hid_composite/hid_composite.ino#L87

@hathach to confirm if that's the API to implement for the wake up.

hathach, 2021-09-22:
@dhalbert `tud_remote_wakeup()` is API to do this, however there is a few requirements, all is implemented by hid example for your reference https://github.com/hathach/tinyusb/tree/master/examples/device/hid_composite
1. `TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP` bit must be set in the configuration descriptor https://github.com/hathach/tinyusb/blob/master/examples/device/hid_composite/src/usb_descriptors.c#L111
2. MCU DCD port remote wakeup and suspend detection must be implemented, which is true for all/most of the port used by circuitpython thanks to recent effort to get them passed USBCV
3. Depending on host OS, before suspending it will send control request to allow device to perform remote wakeup. In my experience, Windows will always allow HID device to perform remote wakeup, but Linux is a bit trickier.  Linux may need explicitly enable, can check which device support remote wakeup with
```
  grep . /sys/bus/usb/devices/*/product
  grep . /sys/bus/usb/devices/*/power/wakeup
```

dhalbert, 2022-01-19:
Bumped priority to do https://forums.adafruit.com/viewtopic.php?f=60&t=187399 and perhaps some other requests I have forgotten.

dhalbert, 2022-03-22:
1. Check descriptor power settings and remote wake
2. Before sending a report, see if we need to wake.

maxammann, 2023-10-16:
Super interesting to discover the lower level USB stuff. I have a Fomu FPGA running CircuitPython. I was able to wake up my screen, but not yet wakeup my device from sleep and I guess it is just not yet implemented.

Is this task just about sending a wakeup message before sending a key press or is it more involved to implement this?

dhalbert, 2023-10-16:
> Is this task just about sending a wakeup message before sending a key press or is it more involved to implement this?

I think we just need to do what is described in https://github.com/adafruit/circuitpython/issues/5380#issuecomment-925044500

PR #8830 (2024-01-25): shared-module/usb_hid: allow HID to wake sleeping host computer
- Fixes #8774.
- Fixes #5380.

Thanks very much to @meesokim for code to enable `usb_hid` to wake a sleeping host computer, and to @hathach for implementing this ability in TinyUSB in the past.

Tested successfully with MacroPad RP2040, Feather nRF52840, and Circuit Playground Express on Windows and macOS.

On Linux not all HID devices are automatically enabled to do wakeup. See @hathach's [comment](https://github.com/adafruit/circuitpython/issues/5380#issuecomment-925044500). See also https://askubuntu.com/questions/848698/wake-u [... cut]
closed 2024-01-25 by dhalbert