adafruit/circuitpython #5391: no module named 'displayio'

Yadro, 2021-09-25:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; Seeeduino XIAO with samd21g18
```

### Code/REPL

```python
import displayio
from digitalio import DigitalInOut, Direction
import board
import time
import adafruit_displayio_ssd1306
import busio

led = DigitalInOut(board.D13)
led.direction = Direction.OUTPUT

displayio.release_displays()

spi = busio.SPI(board.SCL, board.SDA)
tft_cs = board.D2
tft_dc = board.D1
tft_reset = board.D0

display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs, reset=tft_reset, baudrate=1000000)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64)

splash = displayio.Group()
display.show(splash)

inner_bitmap = displayio.Bitmap(118, 24, 1)
inner_palette = displayio.Palette(1)
inner_palette[0] = 0x000000  # Black
inner_sprite = displayio.TileGrid(inner_bitmap, pixel_shader=inner_palette, x=5, y=4)
splash.append(inner_sprite)

while True:
    led.value = True
    time.sleep(1)
    led.value = False
    time.sleep(1)
```

### Behavior

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'displayio'
```

### Description

I want to connect ssd1306 display to Seeeduino XIAO, but facing with import error.

ladyada, 2021-09-25:
displayio is does not fit on the xiao, you can try to use framebuf version
https://learn.adafruit.com/micropython-hardware-ssd1306-oled-display/circuitpython
https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/blob/main/examples/ssd1306_simpletest.py

closed 2021-09-25 by ladyada

Yadro, 2021-09-25:
JFYI. I've tried example above and getting the follow error:
```
Traceback (most recent call last):
  File "code.py", line 6, in <module>
  File "adafruit_ssd1306.py", line 268, in <module>
MemoryError: memory allocation failed, allocating 296 bytes
```
It seems using CircuitPython with ssd1306 is impossible 😢

deshipu, 2021-09-25:
It's not impossible, I use it for example on this device: https://hackaday.io/project/180309-pewpew-oled
But I did have to compile my own version of the firmware, and enable displayio in it.