adafruit/circuitpython #5393: CircuitPython - TinyZero Support

bibletech55, 2021-09-26:
For a personal project I'm working on, I'd like to use CircuitPython as Python is much easier to program in (and I'm much more familiar with CP as well - it worked phenomenally when I upgraded a previous project to use CP!). However, my specific board for this project, the TinyZero, doesn't show CircuitPython support on the website. I would prefer to use this board as it's smaller than a U.S. Quarter, which enables it to fit inside a mini custom enclosure! I believe it's specs are indeed sufficient to run CircuitPython (Atmel 32-bit ATSAMD21G18A ARM Microcontroller, ARM Cortex M0, 256KB Flash, 32KB SRAM), but I couldn't find a compatible .uf2 or .bin file on the CP website for the TinyZero board.

To sidestep the incompatibility, I almost used BOSSA to load MKR Zero files to the TinyZero board as per this tutorial: (https://learn.adafruit.com/welcome-to-circuitpython/non-uf2-installation), but decided against it so far when I discovered MKR Zero code will indeed compile through the Arduino IDE (won't throw an error), but even though no error is thrown, the code won't run (tested with Arduino IDE's Blink; TinyZero's built-in LED only blinks when set to compile to TinyZero, not set to MKR Zero). Honestly, this might indicate the TinyZero is almost compatible with CircuitPython - that only a few things need to be tweaked from the MKR Zero files to add TinyZero support!

With this in mind, would it be possible to add TinyZero support to CircuitPython? It would enable me to use my preferred system of CircuitPython to program my project, and it would also help other creators who need to create extremely miniature projects (such as extremely tiny drones that beginners could program or a miniature accelerometer project). Thank you for your consideration and for all the CircuitPython services already provided! 🙂

dhalbert, 2021-09-26:
See this guide for information about how to add a new board to CircuitPython:
https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

We're happy to get a pull request.

bibletech55, 2021-09-26:
Thank you so very much for your quick reply and for adding this as a long-term update! Have a great day! 😁

dhalbert, 2025-01-31:
Closing due to no progress, but a PR would be welcome.

closed 2025-01-31 by dhalbert