adafruit/circuitpython #5397: Possible conflict of digitalio & pwmio modules on Feather nRF52840 Express' onboard LEDs

tahunus, 2021-09-26:
### CircuitPython version

```python
CircuitPython 6.3 but also observed in version 7.0
```

### Code/REPL

```python
import board
import time

#LEDs-----
import digitalio
redLED = digitalio.DigitalInOut(board.RED_LED)
blueLED = digitalio.DigitalInOut(board.BLUE_LED)
redLED.direction = digitalio.Direction.OUTPUT
blueLED.direction = digitalio.Direction.OUTPUT
blueLED.value = True      #LED is turned on
redLED.value = True       #LED is turned on
time.sleep(2)             #to give me time to see the LEDs

#Buzzer-----
import pwmio
buzz = pwmio.PWMOut(board.A0, duty_cycle=0, frequency=440, variable_frequency=True)   
#this is where blueLED is turned off and becomes unresponsive even to this
blueLED.VALUE = True
time.sleep(3)
```

### Behavior

Both red & blue LEDs onboard the Feather nRF52840 Express are turned on when their led.value is set to True. When the pwm instance is created with PWMOut, the blue led is turned off and becomes unresponsive, even to another led.value=True statement.  The red led is unaffected.

If I invert the sequence and first create the pwm instance and then create the LED instances, the blue LED blinks for an instant and is turned off, even when time.sleep(3) immediately after the led.value=True statement.  Again the red LED is unaffected

Same behavior if executing the code in REPL

dhalbert, 2021-10-08:
I cannot reproduce this problem, using the code above in the REPL. The blue LED stays on when I create the `PWMOut`, and continues to stay on even when I vary the `.duty_cycle`.

Do you have something connected to the `A0` pin, like a buzzer or a speaker? If it's drawing too much current, it can affect the state of other pins. Are there other cthigns connected to the board?

dhalbert, 2021-10-21:
In any case, I am going to close this, but we'll take note of this issue mentally in the future. Thanks for reporting and trying the bootloader update.