adafruit/circuitpython #5400: Crash into HardFault_Handler

axelmagnus, 2021-09-28:
### CircuitPython version

```python
Well, im doing what im told:
I run CP 7.0.0
all libraries are up to date
the content of my CP (Matrix portal M4)
Found device at /Volumes/CIRCUITPY, running CircuitPython 7.0.0.
adafruit_fakerequests==None
neopixel==None
adafruit_requests==None
adafruit_bitmap_font==1.5.1
adafruit_display_shapes==2.3.0
adafruit_display_text==2.21.0
adafruit_esp32spi==3.5.11
adafruit_io==5.5.0
adafruit_matrixportal==3.0.1
adafruit_portalbase==1.9.2

note that the top three doesnt have a version number from Circup.
```

### Code/REPL

```python
4 files: (im sure there are some fonts missing, it'd be a lot easier if i could send you a zipped file with my entire CP on it :) )

secrets.py: (somewhat censored)

secrets = {
    'ssid': 'xxxxx',
    'password': 'xxxx',
    'latitude'  : 55.600520,
    'longitude' : 13.009430,
    'openweather_token' : '4acdd2457856e1ef6c064f1e928ea71e',
    'aio_username' : 'xxxxx',
    'aio_key' : 'xxxxxx'
}

code.py:
import displayio
import adafruit_display_text.label
from adafruit_bitmap_font import bitmap_font
while True:
    from weather_code import weather_code

weather_code.py:

import gc
import time
import board
from adafruit_matrixportal.network import Network
from adafruit_matrixportal.matrix import Matrix
import openweather_graphics  # pylint: disable=wrong-import-position
from secrets import secrets

UNITS = "metric"  # can pick 'imperial' or 'metric' as part of URL query
LOCATION = "Malmö, SE"
print("Getting weather for {}".format(LOCATION))
# Set up from where we'll be fetching data
DATA_SOURCE = (
    "http://api.openweathermap.org/data/2.5/weather?q=" + LOCATION + "&units=" + UNITS
)
DATA_SOURCE += "&appid=" + secrets["openweather_token"]
DATA_LOCATION = []
SCROLL_HOLD_TIME = 0  # set this to hold each line before finishing scroll

# --- Display setup ---
matrix = Matrix()
gc.collect()
[... 300 lines cut ...]
        # Set the position of the group to just off screen and centered vertically for lower half
        self._scrolling_group.x = self.display.width
        self._scrolling_group.y = 24

        # Run a loop until the label is offscreen again and leave function
        for _ in range(self.display.width):
            self._scrolling_group.x = self._scrolling_group.x - 1
            time.sleep(scroll_delay)
        # By blocking other code we will never leave the label half way scrolled
```

### Behavior

Well, infortunately my output from Mu editor is in swedish, but im sure you can translate this standard message. (it goes in to safe mode, and oops, crash in to HardFault_Handler)

Autoladdning Ã¤r avstÃ¤ngd.
KÃ¶r i sÃ¤kert lÃ¤ge! Sparad kod kÃ¶rs inte.

Du Ã¤r i felsÃ¤kert lÃ¤ge eftersom:
CircuitPython kÃ¤rnkod kraschade hÃ¥rt. Hoppsan!
Krasch in i HardFault_Handler.
VÃ¤nligen skapa ett Ã¤rende med innehÃ¥llet i din CIRCUITPY-enhet pÃ¥
https://github.com/adafruit/circuitpython/issues

Tryck pÃ¥ valfri tangent fÃ¶r att gÃ¥ in i REPL. AnvÃ¤nd CTRL-D fÃ¶r att ladda om.

DavePutz, 2022-02-13:
Tested this with CircuitPython 7.2.0-alpha-1 and (after adding the necessary  libs like adafruit_portalbase) everything ran successfully. I changed the frequency to every five minutes and let it run for 1 hour with no issues. @axelmagnus - can you retest with the latest build?

mpicker90, 2022-04-03:
I'm experiencing the same issue calling the OpenWeather API and others running on 7.2.3. reducing the number of API calls makes it run longer on mine but it still ends up crashing eventually. When my Matrix Portal crashes I'm no longer able to set up a serial connection to the device.

dhalbert, 2022-07-06:
Could you all retest again with 7.3.1 or later? We did fix a MatrixPortal-related bug that might have caused this.

DavePutz, 2022-08-05:
Tested this for over an hour on a matrixportal_m4 using CP 8.0.0-alpha.1 with 8.0 libraries and everything worked properly.

dhalbert, 2022-08-06:
@DavePutz - Thanks! I will close this as fixed.

closed 2022-08-06 by dhalbert